node-opcua/node-opcua #820: Bad Node Id Unknown error when read variable but when read Folder or Object is OK

tronghuybku, 2020-06-27:
Im trying to read node ID from Telecontrol Basic Server.
Im getting BadNodeIdUnknown Exception.
I can get type Folder or type Object of Server but when i get VariableNode. it response : BadNodeIdUnknown Exception.
I use Regex Expression to Parse NodeID String when call node id. this variable Node id sample is : 2-TCS:[SeniorStudentHD.Station 1.101]DP,Conveyer_Start with ns = 2 and s = TCS:[SeniorStudentHD.Station 1.101]DP,Conveyer_Start. And i config regex is : ^(\d+)-(?:(\d+)|([\S ]+))$ and this is match.
Hope get your help soon. Thank you

teropes, 2020-06-27:
Hi,
Please provide complete code example here so that people can better understand the issue and try to help. Cheers!

erossignon, 2020-07-01:
@tronghuybku , you code looks like C# (csharp) and not typescript. I wonder if you're posting your question to the right forum
Should it be in https://github.com/OPCFoundation/UA-.NETStandard ? instead

erossignon, 2021-01-19:
Closing due to inactivity

closed 2021-01-19 by erossignon