node-opcua/node-opcua #823: Expecting a valid session here

earlzac50, 2020-07-01:
Hi, everyone.

I faced error about invalid session after I tried to create subscription.

This is code in part of create session.

`
// step 2 : createSessionconst userIdentity = {};
theSession = await client.createSession(userIdentity);
console.log('created session !');
`

erossignon, 2020-07-01:
The session will timeout if there is no communication between the client and the server after the subscription has been created.

Your session doesn't contain any item to monitor and therefore there cannot be any notification send between the client and server.
You need at least one monitor item. Some practice is to monitor server status by default.

earlzac50, 2020-07-02:
@erossignon Thank you for your answer.

I added below code after session created.

```
const browseResult = await theSession.browse('RootFolder');
const nodeValue = await theSession.readVariableValue(nodeId);
```

And added below code after subscription created.

```
  const monitoredItem = ClientMonitoredItem.create(
      theSubscription,
      {
        nodeId: resolveNodeId(nodeId),
        attributeId: AttributeIds.Value,
      },
      {
        discardOldest: true,
        queueSize: 10000,
        samplingInterval: 1000,
      },
    );
    console.log('-------------------------------------');

    monitoredItem.on('changed', (value) => {
      console.log(' New Value = ', value.toString());
    });
```

It still have the same error.

`Error: expecting a valid session here`

erossignon, 2020-07-02:
@earlzac50  You would  need to provide more comprehensive information to help us understand what going on. It is not easy to do so with just partial code extract and an error message without trace. For instance the code to create the client is not exposed and it possible that the default session timeout parameter  has been altered inappropriately.

Ideally, it would be awesome if you could provide a standalone scripts that reproduces the problem if you think this is a bug in node-opcua.

Can you create a small (yet complete)  gist of a simple client script that reproduced the problem ?

earlzac50, 2020-07-02:
@erossignon Sorry for my explanation not clearly.

This is my standalone script that happen an error.

```
const {
  OPCUAClient,
  resolveNodeId,
  AttributeIds,
  ClientMonitoredItem,
  DataType,
} = require('node-opcua');

const client = OPCUAClient.create({
  securityMode: 'None',
  securityPolicy: 'None',

  endpoint_must_exist: false,
  keepSessionAlive: true,

  connectionStrategy: {
    initialDelay: 2000,
    maxDelay: 10 * 1000,
    maxRetry: 10,
  },
});

const endpointUrl = 'opc.tcp://192.168.100.200:4840';
const nodeId = 'ns=4;s=IN200_2';

let theSession = null;
let theSubscription = null;
const run = async () => {
  try {
  // step 1 : connect to
    await client.connect(endpointUrl);
    console.log('connected !');

    // step 2 : createSession
    const userIdentity = {}; // anonymous
    theSession = await client.createSession(userIdentity);
    console.log('created session !');

    theSession.on('session_closed', (statusCode) => {
      console.log("gosh ! session timeout was too short ! let's increase it");
    });

    // step 3 : browse
    const browseResult = await theSession.browse('RootFolder');
    console.log('browseResult: ', JSON.stringify(browseResult, null, 4));
    console.log('browseNames: ', browseResult.references.map((item) => item.browseName));

    // step 4 : read a variable
    const nodeValue = await theSession.readVariableValue(nodeId);
    console.log('nodeValue: ', nodeValue);

    // Test writing variable.
    const nodeToWrite = {
      nodeId,
      attributeId: AttributeIds.Value,
      value: /* DataValue */{
        value: /* Variant */{
          dataType: DataType.Boolean,
          value: false,
        },
      },
    };

    const nodeChanged = await theSession.write(nodeToWrite);
    console.log('nodeChanged: ', nodeChanged);
    console.log('nodeValue 2: ', await theSession.readVariableValue(nodeId));

    // step 5: install a subscription and monitored item
    // //
    // // -----------------------------------------
    // create subscription
    const parameters = {
      maxNotificationsPerPublish: 10,
      priority: 10,
      publishingEnabled: true,
      requestedLifetimeCount: 1000,
      requestedMaxKeepAliveCount: 20,
      requestedPublishingInterval: 2000,
    };

    theSubscription = await theSession.createSubscription2(parameters);

    theSubscription.on('started', () => console.log('STARTED >>>>>> '));

    // install monitored item
    const monitoredItem = ClientMonitoredItem.create(
      theSubscription,
      {
        nodeId: resolveNodeId(nodeId),
        attributeId: AttributeIds.Value,
      },
      {
        discardOldest: true,
        queueSize: 10000,
        samplingInterval: 1000,
        // xx filter:  { parameterTypeId: "ns=0;i=0",  encodingMask: 0 },
      },
    );
    console.log('-------------------------------------');

    monitoredItem.on('changed', (value) => {
      console.log(' New Value = ', value.toString());
    });

    // closing session
    console.log('done!');
    await theSession.close();
    client.disconnect(() => { });
  } catch (error) {
    console.log(' failure ', error);
    client.disconnect(() => { });
  }
};

run();
```

erossignon, 2020-07-16:
try 
```
 theSession = await client.createSession(); // no userIdentity
```
for anonymous connection;

erossignon, 2020-07-16:
The code doesn't wait for the monitored item to be created and goes to the `sesssion.close` too early.

You need let the subscription and monitored item running, therefore wait ( may be until the user press CTRL+C) before gracefully disconnecting.
Also I would use the ClientSubscription#monitor async method to create the monitored item.
This gives
```javascript
const {
    OPCUAClient,
    ClientMonitoredItem,
    resolveNodeId,
    DataType,
    AttributeIds,
    TimestampsToReturn,
} = require("node-opcua");

const client = OPCUAClient.create({
    //  securityMode: 'None',
    // securityPolicy: 'None',

    endpoint_must_exist: false,
    // keepSessionAlive: true,

    connectionStrategy: {
        initialDelay: 2000,
        maxDelay: 10 * 1000,
        maxRetry: 10,
    },
});

const endpointUrl = "opc.tcp://localhost:48010";
const nodeId = "i=2258"; // Server_ServerStatus_CurrentTime";
(async () => {
    try {
        // step 1 : connect to
        await client.connect(endpointUrl);
        console.log('connected !');

        // step 2 : createSession
        const userIdentity = {}; // anonymous
        const session = await client.createSession();
        console.log('created session !');

        session.on('session_closed', (statusCode) => {
            console.log("gosh ! session timeout was too short ! let's increase it");
        });

        console.log("Creating a subscription");
        // create subscription

        const subscription = await session.createSubscription2({
            maxNotificationsPerPublish: 10,
            priority: 10,
            publishingEnabled: true,
            requestedLifetimeCount: 1000,
            requestedMaxKeepAliveCount: 20,
            requestedPublishingInterval: 2000,
        });

        subscription.on('started', () => console.log('STARTED >>>>>> '));

        const itemToMonitor = {
            attributeId: AttributeIds.Value,
            nodeId
        };

        const monitoringParameters = {
            discardOldest: false,
            queueSize: 100,
            samplingInterval: 1000,
        };
        const monitoredItem = await subscription.monitor(itemToMonitor, monitoringParameters, TimestampsToReturn.Both);

        monitoredItem.on('error', (statusCode) => {
            console.log('Error= ', statusCode.toString());
        });
        monitoredItem.on('changed', (value) => {
            console.log(' New Value = ', value.toString());
        });

        console.log('-------------------------------------');
        process.on("SIGINT", async () => {
            // closing session
            await session.close(/* deleteSubscription*/ true);
            await client.disconnect();
            console.log('done!');
            process.exit(0);
        });
    }
    catch (err) {
        console.log("ERROR = ", err);
    }
})();

```

you can set the environment Variable NODEOPCUADEBUG=TRACE before running the program to see the request&response exchange between the client and the server in the console.log.

closed 2020-10-02 by erossignon