node-opcua/node-opcua #825: Monitor Item can return value of not permitted variable

andi00, 2020-07-07:
**Current behavior**
The server returns the value of a blocked variable where the role is not allowed to read.

**Describe the bug**
If you create a monitor item with the samplingInterval 0 the server ignores the Permission of a ua variable 
and return the value even the user role is not permitted.

**To Reproduce**
```typescript
import * as opcua from "node-opcua";
const server = new opcua.OPCUAServer({
  port: 4841,
  allowAnonymous: false,
  userManager: {
    getUserRole(username: string) {
      return username;
    },
    isValidUser(userName: string, password: string): boolean {
      if (userName === "test" && password === "test") {
        return true;
      }
      return false;
    },
  },
});

server
  .start()
  .then(() => {
    const addrSpace = server.engine.addressSpace;
    if (addrSpace === null) {
      return;
    }
    const namespace = addrSpace.getOwnNamespace();
    const options: opcua.AddObjectOptions = {
      organizedBy: addrSpace.rootFolder.objects,
      nodeId: "ns=1;s=Test",
      browseName: "Test",
    };
    const node = namespace.addObject(options);
    const variableOptions: opcua.AddVariableOptions = {
      componentOf: node,
      nodeId: "ns=1;s=TestVariable",
      browseName: "TestVariable",
      dataType: opcua.DataType.String,
      permissions: {
        CurrentRead: ["!*"],
        CurrentWrite: ["!*"],
      },
    };
    const variable = namespace.addVariable(variableOptions);
    variable.setValueFromSource({
      value: "Test",
      dataType: opcua.DataType.String,
    });
    subscripe();
  })
  .catch((reason) => {
    console.error(reason);
  });

function subscripe() {
  const client = opcua.OPCUAClient.create({ endpoint_must_exist: false });
  const user: opcua.UserIdentityInfoUserName = {
    type: 1,
    userName: "test",
    password: "test",
  };
  client.connect("opc.tcp://localhost:4841").then(() => {
    client.createSession(user).then((session) => {
      const subscription = opcua.ClientSubscription.create(session, {
        requestedPublishingInterval: 1000,
        requestedLifetimeCount: 100,
        requestedMaxKeepAliveCount: 10,
        maxNotificationsPerPublish: 100,
        publishingEnabled: true,
        priority: 10,
      });
      const itemToMonitor = {
        nodeId: "ns=1;s=TestVariable",
        attributeId: opcua.AttributeIds.Value,
      };
      const parameters = {
        samplingInterval: 0,
        discardOldest: true,
        queueSize: 10,
      };
      const monitoredItem = opcua.ClientMonitoredItem.create(
        subscription,
        itemToMonitor,
        parameters,
        opcua.TimestampsToReturn.Both
      );
      monitoredItem.on("changed", (value) => {
        console.log(`MonitorItem State: ${value.statusCode.name}`);
        console.log(`MonitorItem Value: ${value.value.value}`);
        client.disconnect();
      });
    });
  });
}
```
**Expected behavior**
The server don't return the current value and return statusCode BadUserAccessDenied

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

First Row is sampling interval 250
Second Row is sampling interval 0

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - (X) Windows : version : Windows 10 Pro
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :

* **node-opcua version:**  : 2.7.0

* **Node:** 
   `node --version` = v12.16.1

erossignon, 2020-07-07:
Well spotted, thank you .

iliareshetov, 2023-09-04:
Hi, I tired to reproduce the issue.

When role permissions properly set I was not able to access the node in question but rather got expected state of **BadUserAccessDenied**. 

Therefore the issue is no longer valid

closed 2024-01-14 by erossignon