node-opcua/node-opcua #833: Can not monitor more than 1000 nodes!?

MasterofDesaster3, 2020-07-22:
I want to subscribe to 1500 nodes with this code:

` for (const node of Nodes) {

      // Create node id of node
      const nodeId = new NodeId(node.identifierType, node.identifier, node.namespaceIndex);
      const itemToMonitor = {
        // Label and attribute of Node
        nodeId: nodeId,
        attributeId: attributeId,
      };

      const monitoredItem = await clientSubscription.monitor(itemToMonitor, parameters, TimestampsToReturn.Both)

      // Listening to Changes of these value (We define Value with AtributeId)
      monitoredItem.on('changed', (dataValue: DataValue) => {
            node.value = dataValue.value.value
      });

    }

    // Return the Subscription
    return clientSubscription; // Return Subscription
  }`

This works with the first 1000 nodes. With node 1001 monitoredItem on "changed" does not trigger.
Is there a workaround?

MasterofDesaster3, 2020-07-22:
Now I put an await to the monitoredItem on function. The maximum is noch 1999 nodes. Is there any chance to increase this to perhaps 10000 nodes to monitor?

mikakaraila, 2020-07-22:
It can be limited by server. I have used 10000 variables with our server.

MasterofDesaster3, 2020-07-23:
Hi, thanks for your reply. The problem was on server side. Thanks! How long does it take you to subscribe to 10000 variables? My Application needs about 6 or 7 minutes. Is that ok?

erossignon, 2020-07-23:
You probably need to subscribe to all your monitored item at once using a monitoring item Group
By doing this, all your monitored item will be created with a single request on the server.

see chapter **9.8 client with large number of monitored items** of [node-opcua by example book](https://leanpub.com/node-opcuabyexample) for a comprehensive example.

closed 2021-01-26 by erossignon