node-opcua/node-opcua #835: Nested Objects

cbaser, 2020-07-23:
Hello,
I want to create a nested objects in server but I can not make it.

`     const addressSpace = server.engine.addressSpace;
        const namespace = addressSpace.getOwnNamespace();

    
        const system= namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "System"
        })

        const controllers = {
            componentOf: system,
            browseName: "Controllers"
        }
        namespace.addObject(controller)

        const controllerX= {
            propertyOf: controllers,
            browseName: "controllerX"
        }

        namespace.addObject(krc)

         namespace.addVariable({
            componentOf: controller,
             browseName: "Manufacturer",
            dataType: "String",
             value: "Example Manufacturer"
         });`

How I can achieve this structure?
Thank you

erossignon, 2020-07-23:
try this

``` 

    const namespace = addressSpace.getOwnNamespace();

    const system = namespace.addObject({
        organizedBy: addressSpace.rootFolder.objects,
        browseName: "System",
    });

    const controllers = namespace.addObject({
        componentOf: system,
        browseName: "Controllers",
    });

    const controllerX = namespace.addObject({
        propertyOf: controllers,
        browseName: "controllerX",
    });

    namespace.addVariable({
        componentOf: controllerX,
        browseName: "Manufacturer",
        dataType: "String",
        value: "Example Manufacturer",
    });
```

cbaser, 2020-07-24:
Hello @erossignon thank you for your help,
I am getting the following error :
`Cannot create property 'dataType' on string 'Example Manufacturer'`
If we can solve it I will close the issue. 
Thank you

erossignon, 2020-07-24:
Opps: I missed that one 
try this:
```
  const manufacturer=  namespace.addVariable({
        componentOf: controllerX,
        browseName: "Manufacturer",
        dataType: "String",
    });
   manufacturer.setValueFromSource({ dataType: "String",     value: "Example Manufacturer" });
```
I recommend the book ["Node-OPCUA by Example"](https://leanpub.com/node-opcuabyexample) to get you up to speed

cbaser, 2020-07-24:
It worked! Thank you very much. I will definitely check the book as well.

cbaser, 2020-07-27:
I am sorry for the re-opening the issue but I need to create a folder inside of the object.

```
const controllersFolder = namespace.addFolder(system,{
        browseName: "ControllersFolder",
    });
```
Can you help me about that as well?
Thank you and sorry for the bothering you.

closed 2020-07-27 by cbaser