node-opcua/node-opcua #838: Connection to OPCUA Server inside Docker not working if not network_mode=host

davimaf, 2020-07-30:
**Current behavior**

Using a OPCUA Server inside Docker container, client connection from UA Expert. 
Image is built from node:12-alpine contianer with folllowing docker-compose file:
```
    edge-mqtt-opcua-server:
        container_name: mqtt-opcua-server
        hostname: edge-mqtt-opcua-server
        image: docker-mqtt-opcua-server:0.0.1
        restart: always
        privileged: true
        #network_mode: host
        ports:
            - 30840:4840
        volumes:
            - edge-mqtt-opcua-server-vol:/data
```
Inside server app, the options for endpoints are:
```
        this.options.port = serverConfig.port;
        this.options.resourcePath = serverConfig.uri;
        this.options.alternateHostname = this.ipAddress;
        this.options.buildInfo = 
        {
            productName: this.hostname,
            buildNumber: "1",
            buildDate: new Date()
        };    
        this.options.serverInfo = 
        {
            applicationUri : "urn:"+ this.hostname + ":" + serverConfig.uri,
            productUri : "node-opcua",
            applicationName : {text: "node-opcua", locale:"en"},
            gatewayServerUri : null,
            discoveryProfileUri : null,
            discoveryUrls : ["0.0.0.0/node-opcua"]
        };
```
If I uncomment **network_mode:host** I can connect UA Expert on opc.tcp://localhost:4840/node-opcua and everythings works fine, but when i remove it (and I need to remove it for this app) I can discover the endpoint with the mapped docker port 30840 but I can't connect UA Expert to Server.
I get the following error in UA Expert:
```
01:58:40.796 | DiscoveryWidget    | | Adding Server node-opcua with URL opc.tcp://edge-mqtt-opcua-server:4840/node-opcua
01:58:43.763 | General            | | [uastack] OpcUa_P_ParseUrl: OpcUa_P_RawSocket_InetAddr reports error 0xA0060000.
01:58:43.766 | DiscoveryWidget    | | Discovery GetEndpoints on opc.tcp://edge-mqtt-opcua-server:4840/node-opcua failed (BadHostUnknown)
01:58:43.769 | General            | | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: ERROR 0x80050000!
01:58:43.769 | General            | | [uastack] OpcUa_SecureConnection_OnNotify: Connect event: Notifying owner! 0x80050000
01:58:43.769 | DiscoveryWidget    | | Discovery GetEndpoints on opc.tcp://localhost:4840/node-opcua failed (BadCommunicationError)
```

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. ....Build a server in docker container
2. ......Map a port to server
3.......DIsable network:mode=host

**Expected behavior**
A clear and concise description of what you expected to happen.

A solution to make reachable OPCUA Server on mapped port and with correct endpoint without use netowork:mode=host that is not usable in some security context.
what I have to modify in my options? I need to do something inside the docker container?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (**X** )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( **X**) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: __**docker**____
  * OS version: ___**node_12-alpine**__
    
    - ( ) Windows : version : _________
    - (**X** ) Linux   : version : _____ubuntu 18.04____
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :
 **2.8.1**
 

* **Node:** 
   `node --version` = 12.18.0

davimaf, 2020-07-31:
By using the same port for mapping on Docker container now is working:
```
  edge-mqtt-opcua-server:
        container_name: mqtt-opcua-server
        hostname: edge-mqtt-opcua-server
        image: docker-mqtt-opcua-server:0.0.1
        restart: always
        privileged: true
        ports:
            - 30840:30840
        volumes:
            - edge-mqtt-opcua-server-vol:/data
```

How I can add a different port to discovered endpoints in server options?

erossignon, 2021-01-23:
Closing due to inactivity

closed 2021-01-23 by erossignon