node-opcua/node-opcua #840: Is it possible to disable the renaming of IDs?

hwsmg, 2020-08-03:
Currently I'm using the node-opcua client (**v2.8.1**) to read values from a PLC and then send them off to other applications. All applications expect the identifiers of the values to match the ID that the OPC server on the PLC has definied. Now, if i try to convert the read value with node-opcua to a JSON object then the identifiers from the PLC get renamed to a camelCase style. This breaks the other applications as they expect the IDs to match the server and I think the OPC UA spec. also tells clients to treat identifiers as case sensetive.

The same issue is present when writing e.g. an extension object. As an example I defined the following variable on my S7-1500 PLC:

[image]

But when I want to write to the "Static_1" variable I have to create the following object:
```
{
    test2: true,
    static_2: {
        teST3: 8,
    },
});
```

If i name the node IDs matching to the PLC like (notice the upper-case S of Static_2 and E of tEST
```
{
    test2: true,
    Static_2: {
        tEST3: 8,
    },
});
```
then mapping the "tEST3" value fails silently. And afterwards I'm writing a 0 to the PLC.

**Describe the solution you'd like**
Preferably a way to disable the modification of IDs by the library completely.  Alternatively at least a way to convert any read value to a javascript object with the correct identifiers as they are specified by the server and the inverse for writing.

teropes, 2020-08-03:
I vote for having it available as configuration. I have had some issues with this as well, even when knowing the behaviour but many things happen programmatically and thus causes issues.

PLC programmers seem to have more variety in variable naming vs the more conventional prpgramming world, they can have spaces in variables and its totally fine in the PLC (at least Siemens i know by experience). Object properties in js can have almost any name / format when accessing with the [] bracket syntax, variable naming is alot more strict. Not sure if the library internally refers to properties with dot syntax relying its valid variable name or if it uses bracket access. Anyways i vote for some control over this default behaviour.

mg90707, 2020-08-04:
> PLC programmers seem to have more variety in variable naming vs the more conventional prpgramming world, they can have spaces in variables and its totally fine in the PLC (at least Siemens i know by experience)

Yes. Also special characters etc. and (unfortunately) the PLC with the server is the "source of truth". Arguing to change all the variable names to a format that conforms to this library will not resonate well with companies that already defined their OPC interface. Also where to stop? Another library might define other rules and then you can't conform to both..

> Not sure if the library internally refers to properties with dot syntax relying its valid variable name or if it uses bracket access.

Wouldn't the library have to be generic - and therefore use bracket notation / indirect access - anyway? I would also guess, that most applications don't have IDs hardcoded into the source, too making the "uniform code style" argument more of a niche case (in my opinion).

@teropes How did you work around the problem? Did you already make a function or something similar to convert the IDs to their original values?

mg90707, 2020-08-04:
As a test I just modified the "lowerFirstLetter"-function, which is responsible for the camel casing, in my node_modules to always return the original string. In my relatively simple example reading and writing still works. The IDs still have their original value in the read data and I can pass an object for writing with the IDs matching the OPC servers definition.

Maybe currently it is really "just for cosmetics" and could be easily disabled or are there some parts of the code that currently depend on that behaviour (or might do so in the future)?

starturtle, 2020-10-24:
This is only referring to string NodeIds, right? In that case I'd also see little sense in manipulating. 

NodeIds are identifiers, and it would surprise me a lot if my server manipulated what I give it as input on its own, so the input string variable I used as NodeId input to create a node just now is not the valid (namespace-specific part of a) NodeId to retrieve it. 
What is the sense of the reformatting in the first place? A NodeId is not a variable name in the coding sense, but an identifier that should not imply anything about the Node names (display or browse) per se. If a server developer chooses to introduce that dependency, then so be it, but that's far from mandatory (what with servers using GUIDs and numeric NodeIds - the namespace would have funny names).

Is there an issue with numeric or GUID NodeIds, too, or do they go unaltered?

mibux, 2021-01-21:
I get similiar issues while reading structured information. I get the names in the value lower camelcased while the structure definition remains like in the PLC.

**Code**
```javascript
const result = await this.session.read({
	nodeId: 'ns=3;s="Steps"."S1"',
	attributeId: AttributeIds.Value,
});
console.dir({ result }, { depth: 6 }); // lower case keys in result.value.value
const typeDescription = await this.session.read({
	nodeId: 'ns=3;s=DT_"UDTSample"',
	attributeId: AttributeIds.DataTypeDefinition,
});
console.dir({ typeDescription }, { depth: 6 }); // typeDescription.value.value.fields[x].name are untouched and the same as on the PLC
```

This was tested on S7-1500 with opc-ua-client v2.30.0, node v12.16.0. I personally woulrd prefer that this remain as sent by the OPC Server.

mg90707, 2021-06-07:
Any update on this? It is really inconvenient for a lot of things and the only upside seems to be that code might look a bit nicer.

david114, 2022-06-09:
The `lowerFirstLetter` Function should be optional. Do I really need to fork this project just to get the original node ID names?

david114, 2022-06-12:
The solution for me was to overwrite the `lowerFirstLetter` Function. If anyone is curious, here's how I did it (quick and dirty): 

`import * as nodeUtils from 'node-opcua-utils';

Object.defineProperty(nodeUtils, 'lowerFirstLetter', {
  value(str : string) {
    return str;
  },
});`

erossignon, 2024-05-10:
Note:
Version of node-opcua@2.120 onward accept JSON objects that takes same PascalCase spelling(original) or camelCase (lowering first letter + some rules).

Combining this with the @sterfive/node-opcua-json module available for subscriber of the node-opcua pro-version and [member of the NodeOCPUA Subscription](https://support.sterfive.com), you get a full support of OPCUA JSON encoding.

closed 2024-05-10 by erossignon