node-opcua/node-opcua #846: Node-OPCUA server returns no values for some data types

Graos, 2020-08-17:
**Current behavior**

**Describe the bug**
For some of the data types server returns NULL value when read from UaExpert client.(Snapshot attached)
**To Reproduce**
Steps to reproduce the behavior:
1. Load the CI874TypesTest.xml nodeset file in OPCUA server written in Javascript using Node-OPCUA 
2. .Read the variable from the IOChannelSet object from UaExpert
3. NULL value is seen in UaExpert for some of the data types and the wireshark also shows no value.

**Expected behavior**
I expect proper initial values for the above said data types. SCreenshots and necessary files attached.

**Gist**
gist:

  my request is related to node-opcua acting as a OPCUA SERVER 
  * ( ) I have installed node-opcua as a package ( using npm install )
 
  * Device: PC
  * OS version: Windows 10
    - ( ) Windows : version : _____10____
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:__UaExpert_______
     - Version:____1.5.1_____
     - Manufacturer/Software vendor:__Unified Automation_______
     - link : https://
     

* **node-opcua version:**  :
 
C:\Users\degarao\Desktop\JSDeplyoment>node OpcUaServer.js
OPC UA Server Application version :  {
  name: 'opcuaserver',
  version: '1.0.0',
  description: 'opcuaserver written in javascript using node opcua framework',
  main: 'OpcUaServer.js',
  scripts: { test: 'echo "Error: no test specified" && exit 1' },
  author: 'Ganesh Rao',
  license: 'ISC',
  dependencies: { 'node-opcua': '^2.10.0' }
}
Node-OpcUa stack version :  ^2.10.0
initialized
Server is now listening ... ( press CTRL+C to stop)
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 1 http://opcfoundation.org/UA/SecurityPolicy#None
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic256
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic256
opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[image]
[SupportingFiles.zip](https://github.com/node-opcua/node-opcua/files/5084595/SupportingFiles.zip)

* **Node:** 
   `node --version` =2.10.0

Graos, 2020-08-17:
I forgot to attach the nodeset file i used to load the OPC UA server. Here it is
[CI874TypesTest.zip](https://github.com/node-opcua/node-opcua/files/5084731/CI874TypesTest.zip)

closed 2020-08-26 by erossignon

julien-schultz, 2020-09-04:
Hi,
I still have this problem, that more abstract variables still have the value null. I also can´t access them with the node-red-contrib-opcua server. Normal variables like String or standard bool are working very fine, but the TwoStateVariableType for example doesn´t work if it is defined in the xml. If I define it by a call in node-red it also works fine. It seems to me, that the variable is not initialized when it´s created by the xml file. Is there any other hint to get these variable types initialized when it´s created by a xml-file? Thanks for your help!
Julien

erossignon, 2020-09-07:
@julien-schultz , I usually initialize those variables explicitly when server is constructed as default value is usually not specified in the xml file. The value at startup is usual domain specific and cannot be guessed nor assumed.

You said : ' TwoStateVariableType for example doesn´t work if it is defined in the xml. '

Can you attach an example of XML nodeset2 file where TwoStateVariable  value is specified ?

julien-schultz, 2020-09-07:
Hi @erossignon, thanks for your quick answer!
In the attachment I have an xml with tree TwoStateVariables. 
TestVariable - already defined True and FalseState in xml
TestVariable2 - a copy of TestVariable with True and False-State as null
TestVariable3 - a independent created TwoStateVariable with all optional components without any changes in UA Modeler.

For your knowledge, I created this XML File in ua Modeler of Unified Automation.

Thanks for your help!

julien-schultz, 2020-09-22:
@erossignon ,
I´ve worked some more time on my Problem with the twoStateVariables.
I remarked, that if I add a twoStateVariable in code, the variable is a UATwoStateVariable.
If it is generated by the XML-Import, it´s only a UAVariable and so the setValue Function is not existing...

Is there a posibility to get it or use it as a UATwoStateVariable from XML Import?

I also bought your book, but this chapter isn´t written yet...

Thank you very much!
Julien Schultz

commit 6539d9fb (2020-08-19): fix #846 extend support for Variable value in nodeset2.xml
commit 33260462 (2020-08-26): fix #846 extend support for Variable value in nodeset2.xml