node-opcua/node-opcua #849: Writing to an Enum Value from external OPC UA client leads to error in node-opcua server

Graos, 2020-08-20:
**Current behavior**

**Describe the bug**
Writing to an Enum variable of ServerState type leads to an write error in node-opcua server. find attched the error log and the nodeset file used to load the OPC UA server.
The variable of focus is EnumTypeData1

  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  
  * ( ) I have installed node-opcua as a package ( using npm install )
 
        
  * Device: _WIndows____
  * OS version: 10_____
    
    - ( ) Windows : version : ___10______
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.11.0
[ErrorLog.zip](https://github.com/node-opcua/node-opcua/files/5101766/ErrorLog.zip)

erossignon, 2020-08-25:
Log file is empty

Graos, 2020-08-25:
Log file Error trace:

C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_modules\node-opcua-factory\dist
\factories_builtin_types.js:253
        throw new Error("datatype " + dataTypeName + " must be registered");
        ^

Error: datatype ServiceCounterDataType must be registered
    at Object.findBuiltInType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_
modules\node-opcua-factory\dist\factories_builtin_types.js:253:15)
    at validateDataType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_module
s\node-opcua-address-space\dist\src\ua_variable.js:114:44)
    at UAVariable._validate_DataType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyomen
t\node_modules\node-opcua-address-space\dist\src\ua_variable.js:1237:16)
    at UAVariable.checkVariantCompatibility (C:\Users\PROFIBUS_USER\Desktop\JSDe
plyoment\node_modules\node-opcua-address-space\dist\src\data_access\ua_analog_it
em.js:23:19)
    at UAVariable.writeValue (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_m
odules\node-opcua-address-space\dist\src\ua_variable.js:512:33)
    at UAVariable.writeValue (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_m
odules\thenify\index.js:65:46)
    at UAVariable.writeAttribute (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\no
de_modules\node-opcua-address-space\dist\src\ua_variable.js:575:22)
    at UAVariable.writeAttribute (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\no
de_modules\thenify\index.js:65:46)
    at ServerEngine.writeSingleNode (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment
\node_modules\node-opcua-server\dist\server_engine.js:919:17)
    at performWrite (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_modules\no
de-opcua-server\dist\server_engine.js:940:18)

Graos, 2020-08-25:
In the attached nodeset file write to variable (EnumTypePropertyData1) leads to the below error in Node OpcUa server

C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_modules\node-opcua-factory\dist
\factories_builtin_types.js:253
        throw new Error("datatype " + dataTypeName + " must be registered");
        ^

Error: datatype SubscriptionDiagnosticsDataType must be registered
    at Object.findBuiltInType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_
modules\node-opcua-factory\dist\factories_builtin_types.js:253:15)
    at validateDataType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_module
s\node-opcua-address-space\dist\src\ua_variable.js:114:44)
    at UAVariable._validate_DataType (C:\Users\PROFIBUS_USER\Desktop\JSDeplyomen
t\node_modules\node-opcua-address-space\dist\src\ua_variable.js:1237:16)
    at UAVariable.checkVariantCompatibility (C:\Users\PROFIBUS_USER\Desktop\JSDe
plyoment\node_modules\node-opcua-address-space\dist\src\data_access\ua_analog_it
em.js:23:19)
    at UAVariable.writeValue (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_m
odules\node-opcua-address-space\dist\src\ua_variable.js:512:33)
    at UAVariable.writeValue (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_m
odules\thenify\index.js:65:46)
    at UAVariable.writeAttribute (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\no
de_modules\node-opcua-address-space\dist\src\ua_variable.js:575:22)
    at UAVariable.writeAttribute (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\no
de_modules\thenify\index.js:65:46)
    at ServerEngine.writeSingleNode (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment
\node_modules\node-opcua-server\dist\server_engine.js:919:17)
    at performWrite (C:\Users\PROFIBUS_USER\Desktop\JSDeplyoment\node_modules\no
de-opcua-server\dist\server_engine.js:940:18)

erossignon, 2020-08-26:
Writing to enum is handled by writin as a basic type;
Here is the test I wrote, and it passes
```
    it("should write a EnumTypePropertyData1 #849", async () => {
        const v1 = propertySet.getPropertyByName("EnumTypePropertyData1") as UAVariable;
        v1.readValue().value.toString().should.eql("Variant(Scalar<Null>, value: <null>)");
        v1.setValueFromSource({
            dataType: "UInt32",
            value: 1,
        });
        v1.readValue().value.toString().should.eql("Variant(Scalar<UInt32>, value: 1)");
    });
````

can you provide a similar test that demonstrate how you attempt to write into the enum variable ?

Graos, 2020-08-27:
Enum variable is written as INT32 please see the attached wireshark capture for the variable NodeId i= 6090
[image]

Graos, 2020-08-27:
For me it appears that the server checks the data type of variable from the nodeset file configuration  and complains about the "ServerState" type not been in the builtin database.

Graos, 2020-08-28:
Even writing the enum variable from UAExpert to node-opcua server results in the same error in the server.

erossignon, 2020-08-28:
OK, I confirm that there is a bug here we need to address.

Graos, 2020-09-09:
When enum type is part of method input argument then "BadInvalidArgument" is returned from node-opcua server. enum is not handled in a right was even the basic data type of the enum argument does not contains the type number of enumeration i.e i=29

Graos, 2021-09-20:
I have a fix for this in my local copy. Can you review the change, Test and merge it ? Let me know I can share the code change with you.

erossignon, 2021-11-24:
fixed in node-opcua@2.60.2

closed 2021-11-24 by erossignon