node-opcua/node-opcua #854: Server returns value of Statuscode variable type always 0x00000000(GOOD) despite forcing any value in server

Graos, 2020-08-28:
**Current behavior**

**Describe the bug**
Server returns value of Statuscode variable type always 0x00000000(GOOD) despite forcing any value in server. This is true even for the method output arguments.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the statuscode type variable in server to say 0x8000000. client receives 0x00000000. I can confirm from the wireshark that the server itself returns 0x00000000 despite the value been initialised to 0x8000000
2. Set the statuscode type output argument of a method in server to say 0x8000000. client receives 0x00000000. I can confirm from the wireshark that the server itself returns 0x00000000 despite the value been initialised to 0x8000000

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: __WIndows___
  * OS version: _10___
    
    - ( ) Windows : version : ____10_____
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:__UAExpert client_______
     - Version:__1.5.1_______
     - Manufacturer/Software vendor:__UA_______
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.12.0

erossignon, 2020-08-28:
Please describe  `Server returns value of Statuscode variable type always 0x00000000(GOOD) **despite forcing any value in serve**`
Also provide sample code and gist to illustrate the way you interact with the server inside your code.

Graos, 2020-08-28:
For instance you can see below the statuscode output argument value been initialized on a method call

function onMethodcallTrigger(inputArguments,sessionContext,callback) {
    console.log("Executing Method = ", this.browseName.name);
    var OutArgs = this.getOutputArguments();
    var OutResults = [];
    for (var i = 0; i < OutArgs.length; i++) {
        OutResults[i] = {};
        if (OutArgs[i].name == "EnumArg") {
            OutResults[i].dataType = opcua.DataType.Int32;
        } else {
            OutResults[i].dataType = OutArgs[i].dataType.value;
        }
        OutResults[i].arrayType = (OutArgs[i].valueRank == -1) ? opcua.VariantArrayType.Scalar:opcua.VariantArrayType.Array;
        switch (OutResults[i].dataType) {
            case opcua.DataType.Boolean:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], BOOL_DEFAULT_VALUE);
                break;
            case opcua.DataType.Double:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], FLOAT_DEFAULT_VALUE);
                break;
            case opcua.DataType.Float:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], FLOAT_DEFAULT_VALUE);
                break;
            case opcua.DataType.Int16:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], INT16_DEFAULT_VALUE);
                break;
            case opcua.DataType.Int32:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], INT32_DEFAULT_VALUE);
                break;
            case opcua.DataType.SByte:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], SBYTE_DEFAULT_VALUE);
                break;
            case opcua.DataType.UInt16:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], UINT16_DEFAULT_VALUE);
                break;
            case opcua.DataType.UInt32:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], UINT32_DEFAULT_VALUE);
                break;
            case opcua.DataType.Byte:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], BYTE_DEFAULT_VALUE);
                break;
            case opcua.DataType.StatusCode:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], STATUSCODE_DEFAULT_VALUE);
                break;
            case opcua.DataType.String:
                OutResults[i].value = getMethodOutputArgumentResultValue(OutArgs[i], STRING_DEFAULT_VALUE);
                break;
            default:
                if (inputArguments[i].dataType == opcua.DataType.ExtensionObject) {
                    OutResults = inputArguments;
                }
                break;
        }
    };
    const callMethodResult = {
        statusCode: opcua.StatusCodes.Good,
        outputArguments: OutResults
    };
    callback(null, callMethodResult);
}

erossignon, 2020-08-28:
Can you provide code for `getMethodOutputArgumentResultValue`

Graos, 2020-08-28:
const STATUSCODE_DEFAULT_VALUE = 0x80000000;
function getMethodOutputArgumentResultValue(OutArg, value) {
    var outValueArray = [];
    var outValue = 0;
    if (OutArg.valueRank != -1) {//Array
        for (let index = 0; index < OutArg.arrayDimensions.length + 1; index++) {
            outValueArray[index] = value;
        }
        return outValueArray;
    }
    else {
        return outValue = value;
    }
}

erossignon, 2020-08-28:
StatusCode's are not number in node-opcua but are handled with a dedicated StatusCode class.

Please try:
```
const STATUSCODE_DEFAULT_VALUE = new ConstantStatusCode(0x80000000,"some description");
```

or more interestingly 
```
const STATUSCODE_DEFAULT_VALUE = StatusCodes.Bad;
```

```StatusCodes is a very handy object that gives you access to all OPCUA available StatusCode defined in OPCUA.
Please refer to the node-opcua-status-code module and its unit test to understand the concept.

Graos, 2020-08-28:
Thanks for the pointers.

erossignon, 2020-10-22:
Using typescript provide type checking during coding time and allows you to spotted such type mismatch more easily.

closed 2020-10-22 by erossignon