node-opcua/node-opcua #861: Namespace.ToNodeset2XML returned error

Graos, 2020-09-11:
Loaded a nodeset file having three namespace. When i tried to call the ToNodesetXML call  on the namespace index 2 callback resulted in exception.

Node-OpcUa stack version :  ^2.14.0
C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:225
                    .join("\n"));
                    ^

TypeError: Cannot read property 'join' of null
    at _dumpVariantValue (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:225:21)
    at _dumpValue (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:414:13)
    at dumpUAVariable (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:672:9)
    at UAVariable.ua_variable_1.UAVariable.dumpXML (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:678:5)
    at dumpAggregates (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:767:23)
    at dumpUAObject (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:735:5)
    at UAObject.ua_object_1.UAObject.dumpXML (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:740:5)
    at dumpAggregates (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:767:23)
    at dumpUAObjectType (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:782:5)
    at UAObjectType.ua_object_type_1.UAObjectType.dumpXML (C:\Users\degarao\Desktop\JSBooks\Project\JavascriptApplication\node_modules\node-opcua-address-space\dist\src\nodeset_tools\nodeset_to_xml.js:786:5)
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. .Load the attached nodeset file and try to call the ToNodesetXML for the namespace index 2.
2. ......
3.......

**Expected behavior**
Expected a string containing the namespace nodeset in xml format
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
    var namespace = server.engine.addressSpace.getNamespace(2);
    var nodestxml = namespace.toNodeset2XML();

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.14.0
[CI874TypesTest.txt](https://github.com/node-opcua/node-opcua/files/5208672/CI874TypesTest.txt)

erossignon, 2020-09-14:
Here is my test program
```typescript
const { OPCUAServer, nodesets } = require("../857/node_modules/node-opcua");
const fs = require("fs");

(async () => {

    const server = new OPCUAServer({
        nodeset_filename: [
            nodesets.standard,
            nodesets.di,
            "./CI874TypesTest.txt"
        ]
    });

    await server.initialize();

    const n = server.engine.addressSpace.getNamespace(2);
    const xml = n.toNodeset2XML();

    fs.writeFileSync("output.xml", xml, "ascii");

    console.log("done");

})();
````
it's happening with empty buffers.

erossignon, 2020-09-16:
should now be fixed in node-opcua@2.15.0 , please confirm

Graos, 2020-09-16:
I will test and confirm. Thanks

Graos, 2020-09-16:
This issue is now resolved in 2.15.0. Thanks for the support.

closed 2020-09-16 by erossignon