node-opcua/node-opcua #865: How the server can be configured to avoid anonymous connection but still using securitypolicy NONE

Graos, 2020-09-16:
**Current behavior**
Server configured with allowAnnonymus false still accepts connection from annonymous client using securitypolicy NONE

**Describe the bug**
Server configured with allowAnnonymus false option still accepts connection from annonymous client using securitypolicy NONE

**To Reproduce**

**Expected behavior**
I expect server to reject the client using anonymous authentication even while using NONE security policy.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
 server = new opcua.OPCUAServer({
        port: CONFIGURATION_SWITCH.SERVER_LISTENING_PORT,        // the port of the listening socket of the server
        nodeset_filename: CONFIGURATION_SWITCH.NODESET_FILENAMES_GENERIC,
         allowAnnonymus:false,
        buildInfo: {
            productName: PRODUCT_NAME,
            softwareVersion: SOFTWARE_VERSION,
            buildNumber: BUILD_NUMBER,
            buildDate: new Date()
        },
        userManager,
    });

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.15.0

erossignon, 2020-10-22:
I've spotted a typo in the code ; it should be `allowAnonymous:false,` instead of `allowAnnonymus:false,`.
Using typescript would help you spot this type of error during coding time.

Graos, 2020-10-22:
Thanks for your excellent support. Now it works. Server does not accept anonymous connection anymore with the above flag set to false.

closed 2020-10-22 by erossignon