node-opcua/node-opcua #870: UAExpert crashes after being rejected by the server with BadCertificateUntrusted

marcodisarno, 2020-09-24:
**Current behavior**

UAExpert crashes when trying to connect to a node-opcua server using a secure channel.

**Describe the bug**
When an OPCUAServer is configured to use Sign&Encrypt with Basic256Sha256 and the client certificate is rejected, UAExpert crashes upon connection attempt.

The **UAExpert** version was **1.5.1 build 331**.

This has been happening on node-opcua version
- **2.5.1**
- **2.12.0**
- **2.16.0**

It has not been seen in
- **2.8.0**

### wireshark analyses
Using the OpcUa protocol on a wireshark capture, the traffic lookes like this:
[image]

##### node-opcua 2.5.11 server return value:
```
OpcUa Binary Protocol
    Message Type: MSG
    Chunk Type: F
    Message Size: 56
    SecureChannelId: 4
    Security Token Id: 0
    Security Sequence Number: 1
    Security RequestId: 1
    OpcUa Service : Encodeable Object
        TypeId : ExpandedNodeId
            NodeId EncodingMask: Four byte encoded Numeric (0x01)
            NodeId Namespace Index: 0
            NodeId Identifier Numeric: ServiceFault (397)
        ServiceFault
            ResponseHeader: ResponseHeader
                Timestamp: Sep 23, 2020 15:19:59.951000000 W. Europe Summer Time
                RequestHandle: 0
                ServiceResult: 0x801a0000 [BadCertificateUntrusted]
                ServiceDiagnostics: DiagnosticInfo
                    EncodingMask: 0x00
                        .... ...0 = has symbolic id: False
                        .... ..0. = has namespace: False
                        .... .0.. = has localizedtext: False
                        .... 0... = has locale: False
                        ...0 .... = has additional info: False
                        ..0. .... = has inner statuscode: False
                        .0.. .... = has inner diagnostic info: False
                StringTable: Array of String
                    ArraySize: 1
                    [0]: StringTable: [OpcUa Empty String]
                AdditionalHeader: ExtensionObject
                    TypeId: ExpandedNodeId
                        EncodingMask: 0x00, EncodingMask: Two byte encoded Numeric
                            .... 0000 = EncodingMask: Two byte encoded Numeric (0x0)
                            .0.. .... = has server index: False
                            0... .... = has namespace uri: False
                        Identifier Numeric: 0
                    EncodingMask: 0x00
                        .... ...0 = has binary body: False
                        .... ..0. = has xml body: False
```
##### node-opcua 2.8.0
```
OpcUa Binary Protocol
    Message Type: ERR
    Chunk Type: F
    Message Size: 53
    Error: 0x80130000 [BadSecurityChecksFailed]
    Reason: An error occurred verifying security.
```

##### node-opcua 2.16.0
```
OpcUa Binary Protocol
    Message Type: MSG
    Chunk Type: F
    Message Size: 80
    SecureChannelId: 9
    Security Token Id: 0
    Security Sequence Number: 1
    Security RequestId: 1
    OpcUa Service : Encodeable Object
        TypeId : ExpandedNodeId
            NodeId EncodingMask: Four byte encoded Numeric (0x01)
            NodeId Namespace Index: 0
            NodeId Identifier Numeric: ServiceFault (397)
        ServiceFault
            ResponseHeader: ResponseHeader
                Timestamp: Sep 23, 2020 16:35:49.627000000 W. Europe Summer Time
                RequestHandle: 0
                ServiceResult: 0x80130000 [BadSecurityChecksFailed]
                ServiceDiagnostics: DiagnosticInfo
                    EncodingMask: 0x00
                        .... ...0 = has symbolic id: False
                        .... ..0. = has namespace: False
                        .... .0.. = has localizedtext: False
                        .... 0... = has locale: False
                        ...0 .... = has additional info: False
                        ..0. .... = has inner statuscode: False
                        .0.. .... = has inner diagnostic info: False
                StringTable: Array of String
                    ArraySize: 1
                    [0]: StringTable: Sender Certificate Error
                AdditionalHeader: ExtensionObject
                    TypeId: ExpandedNodeId
                        EncodingMask: 0x00, EncodingMask: Two byte encoded Numeric
                            .... 0000 = EncodingMask: Two byte encoded Numeric (0x0)
                            .0.. .... = has server index: False
                            0... .... = has namespace uri: False
                        Identifier Numeric: 0
                    EncodingMask: 0x00
                        .... ...0 = has binary body: False
                        .... ..0. = has xml body: False
```

#### findings
The server responds differently depending on the node-opcua version. We found that this correlates with the client crashes like so:

* Server responds with a ServiceFault inside a **Message Type: MSG** -> client crashes while hanging up
* Server responds with a **Message Type: ERR** -> client hangs up correctly
* Also, the error code seems to be worse in newer versions, `BadCertificateUntrusted` was my expection here

**To Reproduce**
Steps to reproduce the behavior:
1. in node-opcua-sample/simple_server.js, set `automaticallyAcceptUnknownCertificate` to `false`
```js
const serverCertificateManager = new OPCUACertificateManager({
    automaticallyAcceptUnknownCertificate: false,
    name: "pki",
    rootFolder: pkiFolder,
});
```
Optionally, force the client to use the security policy

```js
const server_options = {
    securityModes: [MessageSecurityMode.SignAndEncrypt],
    securityPolicies: [SecurityPolicy.Basic256Sha256],
    ...
}
```
2. start the server
3. connect a UAExpert using SignAndEncrypt and Basic256Sha256

**Expected behavior**
UAExpert should abort with error message `BadCertificateUntrusted`.

**Screenshots**
If applicable, add screenshots to help explain your problem.cd ../
[image]

* ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
* (X)  my request is related to node-opcua acting as a OPCUA SERVER 

* (X) I have installed node-opcua from source ( using git clone) (to compare 2.5.11 and 2.8.0)
* ( ) I have installed node-opcua as a package ( using npm install )
* (X) I am using an application that uses node-opcua (2.5.16)
     - (X) node-red
     - ( ) other : please specify 
        
  * Device: PC
  * OS version: Win 10
    
    - (X) Windows : version : 10.0.18363
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
     - Name: UAExpert
     - Version: 1.5.1 331
     - Manufacturer/Software vendor: Unified Automation GmbH
     - link : https://www.unified-automation.com
     

* **node-opcua version:**  : 2.5.11, 2.8.0, 2.16.0

`v2.5.11 = 2b36127fddd546939dd4d85d7fd6d3dc4f3e008d`
`v2.8.0 = 2b36127fddd546939dd4d85d7fd6d3dc4f3e008d`

* **Node:** 
   `node --version` = 12.16.1

marcodisarno, 2020-09-24:
Just tested it with 2.17.0 built from git - problem still exists.

erossignon, 2020-09-28:
Thank you for reporting this,

erossignon, 2020-09-28:
Should be fixed in 2.18.0 , please confirm.

marcodisarno, 2020-09-28:
Yep, it is fixed in 2.18.0. Kudos for reacting so quickly to this!

closed 2020-09-28 by marcodisarno