node-opcua/node-opcua #871: Cannot choose minor version via npm

marcodisarno, 2020-09-24:
**Current behavior**

When trying to pin node-opcua to a minor version (e.g. 2.8.0) does not work. Under the hood, you end up with the most recent release in that minor version.

**Describe the bug**

In our project, we have node-opcua as a dependency and have found some difficulties (see #870). We found that a particular version of node-opcua doesn't exhibit this behaviour, so the solution seemed simple enough: just use that.

#### `package.json`
```json
...
  "dependencies": {
    "node-opcua": "2.8.0"
  },
...
```

But we end up with packages (node-opcua-*) > 2.8.0. This is due to the use of the caret in node-opcua package's [package.json](https://github.com/node-opcua/node-opcua/blob/v2.8.0/packages/node-opcua/package.json).

**To Reproduce**

Steps to reproduce the behavior:
1. `mkdir project-using-node-opcua && cd project-using-node-opcua && npm init`
2. after the prompts, add node-opcua as a dependency (see above) to `package.json`, then `npm install`
3. `grep -A3 -e "node-opcua-server" package-lock.json` to see what the caret did.

**Expected behavior**

End up with version `2.8.0` instead of `2.17.0` or later.

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
 * (X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * () I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * () I am using an application that uses node-opcua 
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: doesn't matter

* **node-opcua version:**  :

* **Node:** 
   `node --version` = **WANTED** `v2.8.0` **but GOT*** `v2.17.0`

marcodisarno, 2020-09-24:
So I know it's a little sassy to put this up as a bug issue, and I get why the carets make absolute sense (updating only one package and all the others benefit, right?).

But consider this: someone might be using this in a productive field, where they have some guarantees to make to their clients. Hard to do when things change without you touching it. So that leaves them no choice but to fork or hardcopy your project, creating a lot of overhead.

You on the other hand seem to be driving this bus forward at great pace - kudos for all the love put into it btw - so it might be an idea to say "Wait for the next release, folks, it's gonna be there soon.", which in fact makes all the carets obsolete.

Maybe you have a good idea or two as to what to do about it. Won't be mad if you just close it or move it to some other forum.

erossignon, 2020-09-27:
This duplicates #765 . This issue has been fixed in 2.10.0. However any previous version cannot be changed as they are already published.
Please update to node-opcua@2.16.0 globally and then minor version update will work.

erossignon, 2020-09-27:
Any reason why you want to pin to 2.8.0 and not upgrade to the latest minor release ? have you found an unwanted regression or breaking change that would need to be resolved ?

marcodisarno, 2020-09-28:
Thanks for the news. Also lerna's `exact: true` is a good thing to know.

Just for the **sake of those who (for whatever reason) depend on a version < `2.10.0`**, I could pin down the version to `2.8.1` using [npm-force-resolutions](https://github.com/rogeriochaves/npm-force-resolutions) (which introduces yarn's [Selective Package Resolutions option](https://classic.yarnpkg.com/en/docs/selective-version-resolutions/) to npm).

A word on this: npm-force-resolutions replaces all the packages in the `package-lock.json` with versions defined in the `package.json`'s `resolutions: {}`.

```json
   ...
   "dependencies": {
        "node-opcua": "2.8.1"
    },
    "resolutions": {
        "node-opcua-address-space": "2.8.1",
        "...and-so-on": "2.8.1"
    },
    ...
```

So the procedure is this:
1. `npm install` the project, including `"node-opcua": "<DESIRED_VERSION_BEFORE_2.10.0>"` as a package using a certain version.
2. check top-level package lerna built: `node_modules/node-opcua/package.json` and copy to `resolutions: {}`
   * all entries in `"depencencies"` with their respective versions (without the caret, of course)
   * all entries in `"_phantomChildren"` as well
3. `npm install` again

That should leave you with every sub-package at versions `<= <DESIRED_VERSION_BEFORE_2.10.0>`.

Pretty fiddly solution, but if you really depend on it, you might want to give this a try.

closed 2020-09-28 by marcodisarno