node-opcua/node-opcua #874: Aggregation giving following Error:  serviceResult = BadAggregateListMismatch (0x80d40000)

shreeramk, 2020-09-28:
Error:  Error:  serviceResult = BadAggregateListMismatch (0x80d40000)

Sample Code:

```
const nodeToRead = [{nodeId: "ns=6;s=2:\\\\osipiserver\\Generator.ABC"}, {nodeId: "ns=6;s=2:\\\\osipiserver\\Generator.CDE"}];

        the_session.readAggregateValue(nodeToRead, "2017-07-31T17:45:00.000Z", "2017-08-03T00:45:00.000Z", 2342, 3600000 , function(err, dataValues, diagnostics) {
            if (err)
                console.log("Error: ", err)
            else {
                console.log('No. of Nodes to be read: ', dataValues.length)
                for (let j = 0; j < dataValues.length; j++) {
                    if (dataValues[j].historyData)
                        console.log("No. of aggregated records fetchted for tag - ", nodeToRead[j], ": ", dataValues[j].historyData.dataValues.length)
                    else
                        console.log(dataValues[j])
                }
            }
        });
```

erossignon, 2020-09-30:
There is not  enough information to understand the issue and reproduce it, I am afraid, please fill the form.

mikakaraila, 2020-09-30:
Or call parameter aggregate function is just wrong? Value 2342 should be 'Average' or some other aggregate.

```javascript
const dataValues = await session.readAggregateValue(
        { nodeId: "ns=5;s=Simulation Examples.Functions.Sine1" },
        "2015-06-10T09:00:00.000Z",
        "2015-06-10T09:01:00.000Z", 'Average', 3600000);
```

shreeramk, 2020-09-30:
@erossignon 
On running the sample code that I mentioned above(Get Aggregate(Average) for two tags between a date range), then I m getting an error which says:

Error: Error: serviceResult = BadAggregateListMismatch (0x80d40000)

Is the full  call stack required here?

erossignon, 2020-10-01:
The error that you see there BadAggregateListMismatch  is generated by the remote server.

@shreeramk the missing information is about the server
- what external server are you trying to connect to ? manufacturer ? version ?
- what version of node-opcua ?
- a full functional gist  (not simply extract)
- are you using typescript of javascript ?
etc...

I guess that you are using javascript and not typescript, therefore there are some type mismatch in passing argument to the method that cannot be detected. Strictly speaking the 

```typescript

    public readAggregateValue(
        nodesToRead: HistoryReadValueIdOptions[],
        startTime: DateTime,
        endTime: DateTime,
        aggregateFn: AggregateFunction,
        processingInterval: number,
        callback: Callback<HistoryReadResult[]>
    ): void;

const dataValues = await session.readAggregateValue(
        { nodeId: "ns=5;s=Simulation Examples.Functions.Sine1" },
        new Date("2015-06-10T09:00:00.000Z"),
        new Date("2015-06-10T09:01:00.000Z"), 
       AggregateFunction.Average, 36000);
````

Note: I realize the sample provided in the code is wrong ... this need fixing

erossignon, 2020-10-01:
comment fixed in 77d46dd1d7062a45b50e1fba6a82f24c04d9374f

erossignon, 2020-10-02:
Comment fix available in 2.19.0 

 @shreeramk  please confirm that this helped

shreeramk, 2020-10-02:
UA Tunneller: Matrikon OPC
node-opcua: 2.14.0 and 2.19.0
Using Javascript.

Summary:
If I try to get aggregated data for one nodeId, it works fine.
_const nodeToRead = [{nodeId: "ns=6;s=2:\\\\osipiserver\\Generator.ACTIVE_POWER"}];_

But when I try to get aggregated data for more than one nodeId then below error is thrown
**Error:  Error:  serviceResult = BadAggregateListMismatch (0x80d40000)**

_const nodeToRead = [{nodeId: "ns=6;s=2:\\\\osipiserver\\Generator.UNIT4_ACTIVE_POWER"},{nodeId: "ns=6;s=2:\\\\osipiserver\\Generator.GEN_CORE_TEMP_3"},];_

erossignon, 2020-10-03:
Ok. I suggest you contact Matrikon support about it. This error is raised by the server itself.

erossignon, 2020-10-04:
@shreeramk  

***BadAggregateListMismatch** means The requested number of Aggregates does not match the requested number of NodeIds 

We need to add the ability to pass an arry of AggregateFunctions when mutilple nodeId are specified.

```typescript
const dataValues = await session.readAggregateValue(
       [
          { nodeId: "ns=5;s=Simulation Examples.Functions.Sine1" },
          { nodeId: "ns=5;s=Simulation Examples.Functions.Sine2" }
      ],
        new Date("2015-06-10T09:00:00.000Z"),
        new Date("2015-06-10T09:01:00.000Z"), 
      [
              AggregateFunction.Average, 
              AggregateFunction.Average
      ],
      36000);
```

shreeramk, 2020-10-04:
@erossignon Yes agreed.

shreeramk, 2020-10-15:
@erossignon can i create a pull request for this?

erossignon, 2020-10-15:
@shreeramk , yes, sure! you're welcome

erossignon, 2020-10-22:
@shreeramk thanks for your contribution, your enhancement is now available in node-opcua@2.22.0

closed 2020-10-22 by erossignon