node-opcua/node-opcua #875: How to read ENUM values from OPC-UA server?

cbaser, 2020-09-29:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - (X) support request 

**Current behavior**

I have created a server code similar to this
 ```
function post_initialize() {
	console.log("initialized");

	function construct_my_address_space(server) {

		const addressSpace = server.engine.addressSpace;
		const namespace = addressSpace.getOwnNamespace();

		// declare a new object
		const device = namespace.addObject({
			organizedBy: addressSpace.rootFolder.objects,
			browseName: "MyDevice"
		});

		const myEnumType = namespace.addEnumerationType({
			browseName: "MyEnumType2",
			enumeration: ["RUNNING", "BLOCKED", "IDLE", "UNDER MAINTENANCE"]
		});
		const enumerationType = addressSpace.findDataType("Enumeration");

		const browseDescription = new BrowseDescription({
			referenceTypeId: null,
			browseDirection: opcua.BrowseDirection.Forward,
			resultMask: 0x3F
		});
		const r = enumerationType.browseNode(browseDescription, null);
		const names = r.map(function (x) {
			return x.browseName.toString();
		});

		const e = namespace.addVariable({
			propertyOf: device,
			dataType: myEnumType,
			browseName: "RunningState",

		});
		e.setValueFromSource({dataType: opcua.DataType.Int32, value: 1});
		console.log(e.readValue())
	}
	construct_my_address_space(server);
	server.start(function () {
		console.log("Server is now listening ... ( press CTRL+C to stop)");
		console.log("port ", server.endpoints[0].port);
		const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
		console.log(" the primary server endpoint url is ", endpointUrl);
	});
}
server.initialize(post_initialize);`
```

And when I am reading this variable I can find the number 1 but I could not find how to react the enumaration value.
Can you help me about that?
Thank you

erossignon, 2020-09-30:
> And when I am reading this variable I can find the nu...

Can you describe how you are "reading this variable " ?

cbaser, 2020-10-01:
Hello,
Firstly thank you for helping me.
I am using the following function to reading a single node variable and I can read 1 after that.

```
async readValue(nodeId) {
       return new Promise((resolve, reject) => {
            the_session.readVariableValue(nodeId, function(err, dataValue) {
                if (!err) {
                    if(dataValue.statusCode._value === 0) {
                        if (dataValue.value) {
                            resolve(dataValue)
                        }
                    }
                    else{
                        console.log("Error reading node with id: " + nodeId + ". Status Code: " + dataValue.statusCode._value + ". Error Description: " + dataValue.statusCode._description)
                        console.log(err)
                        reject(dataValue.statusCode._value)
                    }
                }
                else{
                   console.log("Error Reading "+nodeId + "Error : "+ err);
                    reject(err)
                }

            });
        })

    }
```

erossignon, 2020-10-01:
This is normal the value stored in the enum variables is a integer.

If you want to determine which is the enumeration string that correspond to the value you would have to read the corresponding datatype, read the definition to get either an array of string (0 based indexed) or a map of <integer,string>.

I let the reader to implement it.

Btw, You can greatly simplify you code by not mixing async/await and callback forms. for instance 
```typescript
const dataValue = await the_session.read({ nodeId, attributeId: AttributeIds;Value});
if (dataValue.statusCode !== StatusCodes.Good) {
   ....
} else {
    
```

cbaser, 2020-10-01:
@erossignon thank you for explanation and your suggestion! 
Could you please tell me or give me an example that how to read the datatype? 
Thank you.

erossignon, 2020-10-10:
@cbaser,
Chapter  6.2 Dealing with OPCUA Enumeration types on the server-side and 
Chapter  10.3 reading an enumeration of [Node-OPCUA by example](https://leanpub.com/node-opcuabyexample) provides a comprehensive example to achieve this.

closed 2020-10-22 by erossignon