node-opcua/node-opcua #876: Error: A variant with DataType.ExtensionObject must have a ExtensionObject value

teropes, 2020-09-30:
Hi,

I experienced the following error after doing fresh install of node-opcua 2.6.7. This didn't happen with my previous package-locked installation with same version 2.6.7, it seems some underlying packages have updated and causes this error. I also tested with fresh 2.18.0 installation but same error happens. I can reproduce the error each time by going back and forth between earlier package-lock vs. fresh installs.

Error is
```
(node:29241) UnhandledPromiseRejectionWarning: Error: A variant with DataType.ExtensionObject must have a ExtensionObject value
    at new Variant (/data/s2021-sequencer/sequencer-core/node_modules/node-opcua-variant/dist/variant.js:95:35)
    at new DataValue (/data/s2021-sequencer/sequencer-core/node_modules/node-opcua-data-value/dist/datavalue.js:226:42)
    at ClientSessionImpl.writeSingleNode (/data/s2021-sequencer/sequencer-core/node_modules/node-opcua-client/dist/private/client_session_impl.js:473:20)
    at /data/s2021-sequencer/sequencer-core/node_modules/thenify/index.js:72:10
    at new Promise (<anonymous>)
    at ClientSessionImpl.writeSingleNode (/data/s2021-sequencer/sequencer-core/node_modules/thenify/index.js:70:12)
```

It originates from code that creates array of ExtensionObjects and then writing it using `writeSingleNode` function. This error only happens with newer versions of 2.6.7 installation, not with my previous package-lock installation.

I have grep results from my package-lock related to node-opcua here.

**Working version:**
```
   "node-opcua": {
      "resolved": "https://registry.npmjs.org/node-opcua/-/node-opcua-2.6.7.tgz",
        "node-opcua-address-space": "^2.6.7",
        "node-opcua-address-space-for-conformance-testing": "^2.6.7",
        "node-opcua-assert": "^2.6.1",
        "node-opcua-basic-types": "^2.6.1",
        "node-opcua-binary-stream": "^2.6.1",
        "node-opcua-certificate-manager": "^2.6.5",
        "node-opcua-client": "^2.6.7",
        "node-opcua-client-crawler": "^2.6.7",
        "node-opcua-client-proxy": "^2.6.7",
        "node-opcua-common": "^2.6.5",
        "node-opcua-constants": "^2.6.1",
        "node-opcua-crypto": "^1.3.2",
        "node-opcua-data-access": "^2.6.5",
        "node-opcua-data-model": "^2.6.5",
        "node-opcua-data-value": "^2.6.5",
        "node-opcua-debug": "^2.6.1",
        "node-opcua-enum": "^2.6.1",
        "node-opcua-factory": "^2.6.5",
        "node-opcua-hostname": "^2.6.1",
        "node-opcua-nodeid": "^2.6.1",
        "node-opcua-nodesets": "^2.6.4",
        "node-opcua-numeric-range": "^2.6.5",
        "node-opcua-packet-analyzer": "^2.6.5",
        "node-opcua-secure-channel": "^2.6.5",
        "node-opcua-server": "^2.6.7",
        "node-opcua-server-discovery": "^2.6.7",
        "node-opcua-service-browse": "^2.6.5",
        "node-opcua-service-call": "^2.6.5",
        "node-opcua-service-discovery": "^2.6.5",
        "node-opcua-service-endpoints": "^2.6.5",
        "node-opcua-service-filter": "^2.6.5",
        "node-opcua-service-history": "^2.6.5",
        "node-opcua-service-node-management": "^2.6.5",
        "node-opcua-service-query": "^2.6.5",
        "node-opcua-service-read": "^2.6.5",
        "node-opcua-service-register-node": "^2.6.5",
        "node-opcua-service-secure-channel": "^2.6.5",
        "node-opcua-service-session": "^2.6.5",
        "node-opcua-service-subscription": "^2.6.5",
        "node-opcua-service-translate-browse-path": "^2.6.5",
        "node-opcua-service-write": "^2.6.5",
        "node-opcua-status-code": "^2.6.1",
        "node-opcua-transport": "^2.6.5",
        "node-opcua-types": "^2.6.5",
        "node-opcua-utils": "^2.6.1",
        "node-opcua-variant": "^2.6.5",
        "node-opcua-vendor-diagnostic": "^2.6.7",
[... 1340 lines cut ...]
        "node-opcua-assert": "2.16.0"
    "node-opcua-variant": {
      "resolved": "https://registry.npmjs.org/node-opcua-variant/-/node-opcua-variant-2.17.0.tgz",
        "node-opcua-assert": "2.16.0",
        "node-opcua-basic-types": "2.17.0",
        "node-opcua-data-model": "2.17.0",
        "node-opcua-enum": "2.17.0",
        "node-opcua-extension-object": "2.17.0",
        "node-opcua-factory": "2.17.0",
        "node-opcua-nodeid": "2.17.0",
        "node-opcua-utils": "2.17.0"
    "node-opcua-vendor-diagnostic": {
      "resolved": "https://registry.npmjs.org/node-opcua-vendor-diagnostic/-/node-opcua-vendor-diagnostic-2.18.0.tgz",
        "node-opcua-address-space": "2.17.0",
        "node-opcua-assert": "2.16.0",
        "node-opcua-constants": "2.16.0",
        "node-opcua-server": "2.18.0",
        "node-opcua-status-code": "2.17.0",
        "node-opcua-variant": "2.17.0"
    "node-opcua-xml2json": {
      "resolved": "https://registry.npmjs.org/node-opcua-xml2json/-/node-opcua-xml2json-2.17.0.tgz",
        "node-opcua-assert": "2.16.0",
        "node-opcua-debug": "2.17.0",
        "node-opcua-utils": "2.17.0",
```

erossignon, 2020-09-30:
This behavior is by design, it has been introduced in version 2.13.0.
The exception is thrown to indicate that the code that sets the variant didn't provide a proper extension object.
I suspect this error was not detected in the previous version as it didn't check for the correctness of ExtensionObject.

Looking at the trace, it seems that you are calling writeSingleNode from your client application.

Can you provide the code that preceded this call especially the mechanism by which you are instantiating the ExtensionObject?

erossignon, 2020-10-01:
Also, It appear that you will have to move node-opcua to version >2.11.0 or above regardless as version 2.6.0 suffers from the bug described in #765 . 
My recommandation:
  - update to latest node-opcua@2.17.0 
 - clean node-modules completly and re run npm installl
  - address  your code to fix the issue that causes the exception that detect non Extension object

teropes, 2020-10-01:
Thanks for the help, I was able to pinpoint issue down to our application code. We read array of extension objects in, modify the values of array entries and used to write it back without issues, but with this new validation it failed. I had to ensure that when creating the array for writing, we have to actually construct the extension objects so that they are correct type of objects, plain javascript objects fail the validation.

I got our application working again with latest 2.18.0, so this issue can be closed.

closed 2020-10-01 by teropes