node-opcua/node-opcua #878: Cannot find module 'chalk' in production build

marcodisarno, 2020-10-01:
**Current behavior**
node-opcua-utils does not have "chalk" listed as a dependency

**Describe the bug**
In our production build, that leads to the module chalk not being found by node.

[2020-09-29T12:03:47.827] Error: Cannot find module 'chalk'
Require stack:
- <APP_DIR>\node_modules\node-opcua-utils\dist\compare_buffers.js
- <APP_DIR>\node_modules\node-opcua-utils\dist\index.js
- <APP_DIR>\node_modules\node-opcua-factory\dist\factories_baseobject.js
- <APP_DIR>\node_modules\node-opcua-factory\dist\datatype_factory.js
- <APP_DIR>\node_modules\node-opcua-factory\dist\index.js
- <APP_DIR>\node_modules\node-opcua-data-model\dist\access_level.js
- <APP_DIR>\node_modules\node-opcua-data-model\dist\index.js
- <APP_DIR>\node_modules\node-opcua-types\dist\_generated_opcua_types.js
- <APP_DIR>\node_modules\node-opcua-types\dist\index.js
- <APP_DIR>\node_modules\node-opcua-common\dist\index.js
- <APP_DIR>\node_modules\node-opcua\dist\index.js

Our workaround is to add chalk as a dependency of our top-level app, which of course, we'd get rid of.

**To Reproduce**
Sadly, this could not be reproduced in a minimal example, so it maybe something related to our build. It uses lerna 2.9 to build and publish packages to a private npm registry. 

**Expected behavior**
Get a working production build.

 * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
 * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

 * ( ) I have installed node-opcua from source ( using git clone)
 * (x) I have installed node-opcua as a package ( using npm install )
 * ( ) I am using an application that uses node-opcua 
        
  * Device: PC
  * OS version: Windows 10
    
    - (X) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
* **node-opcua version:**  : 2.18.0

* **Node:** 
   `node --version` = 2.16.1

marcodisarno, 2020-10-01:
This would solve the problem:

```diff
--- packages/node-opcua-utils/package.json
+++ packages/node-opcua-utils/package.json
@@ -11,11 +11,11 @@
     },
     "dependencies": {
         "browser-process-hrtime": "^1.0.0",
-        "chalk": "^4.1.0",
         "node-opcua-assert": "2.16.0"
     },
     "devDependencies": {
         "@types/node": "^14.11.1",
+        "chalk": "^4.1.0",
         "should": "^13.2.3",
         "sinon": "^9.0.3"
     },
```

erossignon, 2020-10-02:
Fixed in version 2.19.0,

closed 2020-10-02 by erossignon