node-opcua/node-opcua #880: Nano profile for OPC UA Server using node-opcua

msalafia, 2020-10-02:
What is the right procedure to define an OPC UA Server supporting the Nano Embedded profile only? I cannot find samples or other issues showing how to do that. What i wan to achive is developing a lightweight server on a constrained device.
Furthermore, i would like to know whether such Nano Embedded-compliant Server would be deployed on the constrained device with a reduced npm package or not.

erossignon, 2020-10-02:
We took the approach "Who can do more can do less"

You would have to disable some functionality during the configuration of the server
 - Limit the number of concurrent connections to 2 ( instead of 50 by default)
 - only enable unsecured connection (by specifying endpoints, instead of using the default full-blown endpoint set)
 - set up to sensible server limited (operational limits etc ... )
 - load a lighter standard nodeset file so it doesn't come will all the bells and whizzles.
 - overwrite unneeded service to return BadUnsupported instead of processing the data (such as Subscription /Monitored Item, Transfer ,  Method call etc ... )
 - disable auditing 
 - etc ... 

This will allow you to limit the resources used by the server at run time (main memory consumption) but probably no the size of the runtime program (which comes with every feature built-in). The memory footprint will be in the region of 140mb for comfortable use.

If you need to work on really smaller devices I would recommend to use the C/C++ stack open62541 from our Fraunhofer colleagues.

What is the spec of the computer board you have in mind to run the nano embedded server ?

msalafia, 2020-10-02:
Thank you for your fast answer @erossignon . I have no bord in mind, i was just wondering which was th approach to a nano server with node-opcua eventually if in future i need to implement one. Your answer is more then satisfying. 

Just one question related to the information you provided me: what you mean (in practical) with "overwrite unneeded service to return BadUnsupported instead of processing the data" ? Should i change all the method relevant to services? If yes, which packages/files? Of course i dond wont you list me all files or packages :D just want to know if there is some logics or philosophy behind node-opcus services implementation so that i can understand which method and files edit.

erossignon, 2020-10-02:
https://apps.opcfoundation.org/ProfileReporting/

Here is the logic, you could use , something like

```
class OPCUANanoServer exends OPCUAServer {

  protected _on_DeleteReferences(message: Message, channel: ServerSecureChannelLayer) {
          return g_sendError(channel, message, DeleteReferencesResponse, StatusCodes.BadNotImplemented);
   protected _on_CreateSubscription(message: Message, channel: ServerSecureChannelLayer) {
          return g_sendError(channel, message, CreateSubscriptionResponse, StatusCodes.BadNotImplemented);
 }
  etc ....   
}
```

msalafia, 2020-10-02:
Thank you very much!

closed 2020-10-02 by msalafia