node-opcua/node-opcua #881: Can't connect with alternateHostname

andi00, 2020-10-05:
**Current behavior**
If i try to connect with a login to the alternateHostname ip the server anwsers with a BadIdentyTokenInvalid

**Describe the bug**
The bug is that you can't use the alternateHostname of the server to connect to him.
It looks like at the activateSession the wrong endpoint on the server is used. It use the default endpoint.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Code belowe
2. Try to connect with UaExpert

import * as opcua from 'node-opcua';
const server = new opcua.OPCUAServer({
   port: 4841,
   allowAnonymous: false,
   userManager: {
      isValidUser(userName: string, password:string): boolean {
            if(userName === "test" && password === "test"){
                return true;
          }
         return false;
    }
  },
 alternateHostname: "10.9.1.63"
});

server.start().then(() => { 
      console.info('server started');
}).catch( err => {
    console.error(err);
});

**Expected behavior**
I can connect to the server with the alternateHostname ip.

**Screenshots**

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (X)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (X) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  : 2.19.0

* **Node:** 
   `node --version` = 13.14.0

erossignon, 2020-10-06:
Thank you, I can reproduce;

erossignon, 2020-10-11:
@andi00, can you test with node-opcua@2.20.0 ?  it should be fixed now. please confirm

andi00, 2020-10-12:
Hallo Etienne, 
i think have now two other problem if i try to install the 2.20.0 or 2.19.0 in my typescript projekt and try to built it with tsc i get errors in the node-opcua-crypto lib. I get the message 'This expression is not callable' at lines with 'assert(.... '.

On my docker build where i only use javascript the service know works with the alternateHostname but not with localhost or 127.0.0.1 what he has done with 2.19.0.

erossignon, 2020-10-12:
what is the version of  node-opcua-crypto installed on your side ?

erossignon, 2020-10-12:
> On my docker build where i only use javascript the service know works with the alternateHostname but not with localhost or 127.0.0.1 what he has done with 2.19.0.

* You need to access the server with one of the endpoint uri provided by the server.  Alternatively , try not to specify alternateHostname all together;

andi00, 2020-10-12:
> of

According to the package-lock.json the node-opcua-crypto version is the 1.4.2.
If i expliced install

andi00, 2020-10-12:
If i set the version from node-opcua-pki fix to 2.1.7 and node-opcua-crypto to 1.3.8 i can build the typescript project.

erossignon, 2020-10-12:
I have issued node-opcua-crypto@1.4.3 that shall revert some changes regarding how to import "assert" packages... Alternatively you could adjust the     `"esModuleInterop"` flag in your `tsconfig.json` file.

Alia5, 2020-10-13:
> Alternatively you could adjust the "esModuleInterop" flag in your tsconfig.json file.

This sadly is really bad, we rely on `esModuleInterop` in our project... And now its broken

erossignon, 2020-10-13:
@Alia5  so use the other alternative and make sure node-opcua-crypto@1.4.3  is installed,
only node-opcua-crypto@1.4.1 had the regression.

erossignon, 2020-10-16:
@Alia5, in fact you'll need node-opcua-crypto@1.4.4 patch ...

closed 2020-10-16 by erossignon