node-opcua/node-opcua #884: OPC UA client (UAExpert) is not connecting to Node-OPCUA server anymore (from version 2.20.0)

Graos, 2020-10-13:
**Current behavior**
OPC UA client (UaExpert) is not connecting to Node-OPCUA server anymore (from version 2.20.0). Until 2.19.0 connection works fine. I tried all the security policy. Everything results in same issue.

**Describe the bug**
OPC UA client (UaExpert) is not connecting to Node-OPCUA server anymore (from version 2.20.0). Until 2.19.0 connection works fine. I tried all the security policy. Everything results in same issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Node-OPCUA server
2. .Start UAExpert client and connect to Node-OPCUA server with any security policy. I have attached the log for None Security policy. The same works with 2.19.0 Node-OPCUA

**Expected behavior**
OPC UA client should connect to Node-OPC UA server for all the security policy options.
**Screenshots**
UAExpert Log snippet:
***********************************************************************************************
10:45:39.788 | Server Node        | JSOpcUaServer                  | Endpoint: 'opc.tcp://DE-L-7224240.europe.abb.com:26543'
10:45:39.788 | Server Node        | JSOpcUaServer                  | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
10:45:39.788 | Server Node        | JSOpcUaServer                  | ApplicationUri: 'urn:NodeOPCUA-Server-default'
10:45:39.788 | Server Node        | JSOpcUaServer                  | Used UserTokenType: Anonymous
**10:45:42.303 | General            |                                | Error: UaSessionPrivate::activateSession - no endpoint matching the session security settings
10:45:42.305 | Server Node        | JSOpcUaServer                  | Error 'BadConfigurationError' was returned during ActivateSession**
10:45:42.311 | Server Node        | JSOpcUaServer                  | Connection status of server 'JSOpcUaServer' changed to 'Disconnected'.
*********************************************************************************************

Wireshark capture
[image]

**Gist**
gist:

  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua as a package ( using npm install )

        
  * Device: _Windows____
  * OS version: ___10__
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:___UAExpert______
     - Version:___1.5.1______
     - Manufacturer/Software vendor:__Unified Automation_______
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.20.0

erossignon, 2020-10-13:
in 2.20.0 endpoint matching has been reenforce and client must access server with  a known endpoint :

Can you please run this [gist script](https://gist.github.com/erossignon/88d73c43bba876012800e0f1c07163ea#file-getendpointbasic-js
) and provide the edndpoints exposed by your server ?

Graos, 2020-10-13:
Ok. opc.tcp://DE-L-7224240.europe.abb.com:26543 is the endpoint exposed by the server and the end point configured at OPC UA client is opc.tcp://DE-L-7224240.EUROPE.ABB.COM:26543

The difference i see is the change in the case. I changed  the endpoint URL at the client side to match exactly the same as the server and it works. Thanks for the support.

erossignon, 2020-10-13:
OK, I have also [updated the release note](https://github.com/node-opcua/node-opcua/releases/tag/v2.20.0)  as other users might be trapped by this new (but more conformant) behavior.

closed 2020-10-13 by erossignon