node-opcua/node-opcua #886: Cannot compile typescript when esModuleInterop is true

codanator, 2020-10-14:
Ciao,

today i installed node-opcua 2.21.0 and typescript 4.0.3.
Now i cannot compile the project because node_modules/node-opcua-crypto/source/asn1.ts errors @ assert()
node-opcua-crypto version 1.4.3 is installed.

When i disable esModuleInterop it compiles!

erossignon, 2020-10-15:
see also https://stackoverflow.com/questions/56238356/understanding-esmoduleinterop-in-tsconfig-file

Wensh, 2020-10-16:
Disabling esModuleInterop is giving me a lot of issues with other packages moment, Aedes etc. And migrating everything to start_import is breaking all my Jest testing mocks. 

@erossignon esModuleInterop is gone for good with node-opcua?

erossignon, 2020-10-16:
Sorry guys, It was not intentional. 
I simply don't get it. I see also the behavior on my end on some projects as well. 
Help would be appreciated to find the root cause of this ...

erossignon, 2020-10-16:
I think I nailed it ...  I accidentally imported the full typescript code of the node-opcua-crypto modules instead of just exposing the compiled typed definitions. I'm issuing a patched version of node-opcua-crypto now.. 🐻 with me.

erossignon, 2020-10-16:
OK, you can fix it now by installing node-opcua-crypto@1.4.4 explicitly alongside with node-opcua.
A permanent fix will be issued in the next minor release of node-opcua.

erossignon, 2020-10-22:
fixed in node-opcua@2.22.0

closed 2020-10-22 by erossignon