node-opcua/node-opcua #894: Question: node-opcua Server exposed Endpoints

AndreasHeine, 2020-10-26:
@erossignon long time no see :)

i am working on an opcua server which only exposes featured endpoints. 

maybe i missundestood the docs but i was thinking that if i set the serveroptions like
```js
    securityMode: [opcua.MessageSecurityMode.None],
    securityPolicy: [opcua.SecurityPolicy.None],
```
only these endpoint will be exposed?

http://node-opcua.github.io/api_doc/2.0.0/interfaces/opcuaserveroptions.html#securitymodes

[image]
[image]

lg Andreas

erossignon, 2020-10-26:
Almost there ! 
you should be using `securityModes` and `securityPolicies` instead !

In order to avoid this sort of pitfall, you can either use Typescript or add the triple slash directive at the top of your scrip file.

In this case, Intellisense will work in VSCode:

[image]

AndreasHeine, 2020-10-26:
... ok fail ... (should have listen to you months ago about typescript -_-° return new Promise((resolve, reject) =>{ //typescript ;) })

```js
const opcua = require("node-opcua");

const server = new opcua.OPCUAServer({
    port: 4840,
    hostname: "192.168.3.103",
    resourcePath: "/UA",
    buildInfo : {
        productUri: "MES 1",
        productName: "MES",
        manufacturerName: "Andreas Heine",
        buildNumber: "v1.0.0",
        buildDate: new Date(2020,10,26)
    },
    userManager: userManager,
    allowAnonymous: false,
    securityModes: [opcua.MessageSecurityMode.None],
    securityPolicys: [opcua.SecurityPolicy.None],
    disableDiscovery: true,
});
```

thanks Etienne!

closed 2020-10-26 by AndreasHeine