node-opcua/node-opcua #895: Close ClientSession from OPCUA server

icemini, 2020-10-27:
I'm facing an issue with the termination of the OPCUA client session. 

The current behavior:
The client session still exist on the OPCUA server and is not closed properly until a timeout is received and thereby affecting the maxAllowedSessionNumber count on my server even though the client has been terminated.

What I do:
Both my OPCUA server and client are running and I terminate the client with a CTRL+C command in the terminal. 

Problems:
When I do CTRL+C on my client the server does not receive a "session_closed" event, however it does receive a "closeChannel" event. From this channel I am able to extract the ServerSession and closing it and receiving a session_closed event on the server. But this still does not terminate the ClientSession and keeps it open.

`server.on("closeChannel", function (channel) {

//extract ServerSession and close it
            Object.values(channel.sessionTokens)[0].close(true, "Terminated");

        });`

Is there anyway I can manually close the ClientSession from the server side when it has been closed with CTRL+C on the client?

Thank you.

erossignon, 2020-10-27:
Session must be kept alive for the duration of timeout to allow a possible reconnection by a resuming client. The closeChannel may have been received because of a network outage. So the server has no idea why the connection has dropped. 

Here are some ideas:

- You can add code to intercept the "CTRL+C"  in your client and perform a proper session close and client disconnection before terminating.
- You can increase the number of allowed connection on your server (10 is the default, 500 could be used)
- You can decrease the timeout of your session in the OPCUAClient constructor so that unused sessions will be terminated faster on the server-side.

Some examples of how this can be accomplished can be found in the [node-opcua by example](https://leanpub.com/node-opcuabyexample)  book.

icemini, 2021-01-25:
Thank you, this cleared up a lot of things for me!

closed 2021-01-25 by erossignon