node-opcua/node-opcua #897: Timeout writing to a node, server get the request but not response back

vadhack, 2020-10-28:
Hello, someone know what am i doing wrong? i write to a node from client opc and the server get the request but never response and then 1 minute the resquest end up as Timeout

some Server code:

```coffeescript
server = new opcua.OPCUAServer
    port: 4840
    resourcePath: "/"
    buildInfo : 
        productName: "IELCOOPC"
        buildNumber: "1"
        buildDate: new Date()

...

node = namespace.addVariable
    browseName  : ns
    componentOf : device
    dataType    : type
    nodeId      : nodeId
    value: 
        get: () =>
            value = if variable.node then variable.node._dataValue.value.value else utils.get_value_type(type)
            return new opcua.Variant 
                dataType: type 
                value: value 

        set: (variant, cb) =>
            ENABLE_READ = false

            plc.write(variant.value, add.address, type)
            .then ([err, value, code]) =>
                if not err and value
                    variable.node.setValueFromSource({ dataType: variable.type, value: value}) 
                ENABLE_READ = true

            variable.node.setValueFromSource(variant) 
            ENABLE_READ = true
            return opcua.StatusCodes.Good

```

some Client code:

```coffeescript

client  = opcua.OPCUAClient.create
    endpoint_must_exist:false
    keepSessionAlive: true
    securityMode: MessageSecurityMode.None
    securityPolicy: SecurityPolicy.None

port_ = 4840
hostname    = '127.0.0.1' #'10.10.100.1' # '127.0.0.1'
endpointUrl = "opc.tcp://#{hostname}:#{port_}/"
client.connect endpointUrl, (err) =>
    return onconnect(err) if err

    ses =
        connectionStrategy: 
            maxRetry: 10
            initialDelay: 2000
            maxDelay: 10*1000

    client.createSession ses, (err,session) =>
        return onconnect err if err
        subscription = opcua.ClientSubscription.create session,
            #requestedPublishingInterval: 100
            #requestedLifetimeCount: 10
            #maxNotificationsPerPublish: 10
            publishingEnabled: true
            priority: 1

...

params = 
    nodeId: nodeId
    attributeId: opcua.AttributeIds.Value
    indexRange: null
    value: 
        value: dataToWrite

console.log "writing..."
res = await session.write params
console.log "NEVER RESPONSE"

even with callbacks:

session.writeSingleNode nodeId,dataToWrite, (err, statusCode, diagnosticInfo) =>
    console.log "NEVER RESPONSE"

```
i use the same client code for connect to KEPSERVER OPC Server and i can write to nodes successfully but not with a Node-OPCUA server

Hope you understand my problem and can help me, Thank you

**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - (x ) support request 

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

* **Typscript:** 
   `tsc --version` =

vadhack, 2020-10-29:
the request end with this:

> Request has timed out without receiving a response: WriteRequest {
  requestHeader: RequestHeader {
    authenticationToken: NodeId {
      identifierType: 4,
      value: <Buffer 5c 50 9b 6a 80 7d 60 0e 7f d7 fa d3 0a c5 71 9b>,
      namespace: 0
    },
    timestamp: 2020-10-29T16:16:55.051Z { high_low: [Array], picoseconds: 0 },
    requestHandle: 6297,
    returnDiagnostics: 0,
    auditEntryId: '',
    timeoutHint: 60000,
    additionalHeader: null
  },
  nodesToWrite: [
    WriteValue {
      nodeId: [NodeId],
      attributeId: 13,
      indexRange: [NumericRange],
      value: [DataValue]
    }
  ]
}

erossignon, 2020-10-31:
you forgot to call the `cb`callback in the set lambda

vadhack, 2020-11-03:
Hello, thanks, i called "cb" callback before but also not responding

erossignon, 2020-11-03:
May be plc.write never returns. 
What sort of esoteric JavaScript dialect are you using ? 
May be TypeScript would help you figure out that set do not take a callback.
Assuming that plc.write is async it seems that the setter returns before the value is written in the plc. Please use the async form of the variable setter as described in the book.

vadhack, 2020-11-09:
Hello, thanks for the help, 

I comment you: I found an error in the client code: I wrote to a node with a wrong DataType, when I fixed the error of the DataType (in the client code) the server began to response correctly every time I wrote, so I assume that it was not a problem from my server code because I did call the "cb" callback or return "opcua.StatusCodes.Good" in the "set" method.
I can deduce that there could be a data type interpretation error (on the server) that is not caught when it occurs and that is why a Timeout error is thrown after 1 minute.
what seems strange to me is that the "set" method does execute (also plc.write) but then the server no longer responds.

in summary: if I write to a node from the client with a wrong DataType the server responds with Timeout.

that was what I observed. 

I use Coffeescript language.

Thanks.

closed 2020-11-18 by vadhack