node-opcua/node-opcua #898: User has multiple roles

MarcoSeb, 2020-10-29:
# Feature Request
The OPC Server does not support a user having multiple roles. This is a very common use case.

# Problem
Using LDAP-user to grant Permissions to the device.
Every user is assigned multiple groups(opc: _Roles_). Every group grants permission to _Nodes_ in the _AddressSpace_

# OPC Spesification
[The OPC UA approach assigns Permissions to Roles for each Node in the AddressSpace. Clients are then granted Roles when they create a Session based on the information provided by the Client](https://reference.opcfoundation.org/v104/Core/docs/Part5/F.1/)

[The OPC UA approach assigns Permissions to Roles. Clients are then granted Roles based on connection information](https://reference.opcfoundation.org/v104/Core/docs/Part2/4.12/)

# Solution
`export type GetUserRoleFunc = (username: string) => string;`
should allow to return multiple roles
`export type GetUserRoleFunc = (username: string) => string[];`
For backwards compatibility:
`export type GetUserRoleFunc = (username: string) => string | string[];`

erossignon, 2020-11-02:
you can combine multiple roles by separating them with `;' for instance `AuthenticatedUser;SecurityAdmin`.

MarcoSeb, 2020-11-03:
Thanks, I started to work on a PR and realize that is already possible.
This feature is not obvious from the interface.

For now, it should be explained in the doc.

erossignon, 2020-11-04:
available in node-opcua@2.24.0

closed 2020-11-04 by erossignon