node-opcua/node-opcua #899: node-opcua server does not initialize the NodeId variable with its value configured in nodeset file.

Graos, 2020-10-30:
**Current behavior**
ExclusiveLimitAlarm object instance "Input node" variable component is configured with the NodeId of the eventsource (FloatTypeData1). After the nodeset file is loaded into node-opcua server the "Input node" variable component is not initialized with the NodeId value configured in the nodeset file. The value type is shown as null in UAExpert.

**Describe the bug**
ExclusiveLimitAlarm object instance "Input node" variable component is configured with the NodeId of the eventsource (FloatTypeData1). After the nodeset file is loaded into node-opcua server the "Input node" variable component is not initialized with the NodeId value configured in the nodeset file. The value type is shown as null in UAExpert. Because of this Alarm and event generation does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the attached nodeset file into the server
2. Read the Input node variable of the ExclusiveLimitAlarm object configured under IOChannelSet object.
3.The value is empty and the type is shown as null

[image]

**Expected behavior**
I expect the alarm and event to work.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Gist**
gist:

 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

 
  * ( ) I have installed node-opcua as a package ( using npm install )
 
 
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:____UaExpert_____
     - Version:______1.5.1___
     - Manufacturer/Software vendor:___Unified Automation______
     - link : https://
     

* **node-opcua version:**  :

* **Node:** 
   `node --version` = 2.22.0

Graos, 2020-10-30:
Nodeset file attached.
[CI874TypesTestAlarm.zip](https://github.com/node-opcua/node-opcua/files/5463199/CI874TypesTestAlarm.zip)

Graos, 2020-11-02:
I am not sure if this is a support request. I thought it is an issue with the Node-opcua server.

erossignon, 2020-11-02:
It may be an issue but nevertheless, this is an unimplemented feature in the product. 
It's a support request as you are requesting external people to write the required enhancement for you. 
Our business model is to improve node-opcua for all. However, we do need the support of our users.
There are several ways to contribute:
   - ask the community, 
   - jump in and write a PR  yourself and get the author helping you to integrate 
   - ask & encourage  the author by (you or your employer ABB) becoming [a sponsor](https://github.com/sponsors/node-opcua)
   - or request professional support and establish a sustainable relationship with the author and possibly influence the direction of the roadmap. 
   
This being said the parsing of "NodeID"  value has been added especially for you on node-opcua@2.21.0 ....  ball's on your side.

Graos, 2020-11-02:
OK Thanks. I agree.

closed 2020-11-02 by erossignon