node-opcua/node-opcua #902: Error("Parent folder must be of FolderType " + folder.typeDefinition.toString());

AndreasHeine, 2020-10-30:
Hey Etienne,

I try to organize my addressspace a little and i am getting a confusing error i want to create a folder as children of an BaseObjectType-Node similar to:

[image]

but i get:
> throw new Error("Parent folder must be of FolderType " + folder.typeDefinition.toString());
> Error: Parent folder must be of FolderType ns=0;i=58

```js

        const addressSpace = server.engine.addressSpace;
        const namespace = addressSpace.registerNamespace("http://testnamespace/ua/");

        const myNode = namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "myNode"
        });

        const myfolder = namespace.addFolder(
            myNode,
            {
            browseName: "myFolder"
            }
        );
```

lg Andreas

erossignon, 2020-10-31:
Hi Andreas,

You would need to  use a different way: 

````javascript

    const namespace = addressSpace.getOwnNamespace();

    const myFolder = namespace.addObject({
        browseName: "MyNewFolder",
        organizedBy: addressSpace.rootFolder.objects.server,
        typeDefinition: "FolderType",
    });

````

erossignon, 2020-10-31:
The reason why addFolder restricts parent object to be of type FolderType is due to this statement in the OPCUA Specification:
[image]

The SourceNode of such a Reference should always be a View or an Object of the ObjectType FolderType; the TargetNode can be of any NodeClass.

So strickly speaking "MyNode" must be of type "FolderType"

````
   const myNode = namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "myNode"
             typeDefinition: "FolderType",
        });
````

closed 2020-11-04 by AndreasHeine