node-opcua/node-opcua #904: Invalid Server Nonce

mirkocomparetti-synesis, 2020-11-02:
**Current behavior**
When connecting to an OPC-UA server using anonynmous authentication (None/None for security policies, ...) it is possible that the server responds with an invalid server Nonce, thus preventing connection to the server.

**Describe the bug**
Now in case of Anonymous authentication, the connection using the node-opcua client leads to `[FATAL] main - Error: Invalid server Nonce`

**To Reproduce**
1. Create an opcua server that has anonymous authentication and sends null as ServerNonce for authonomous authentication.
2. connect to the server using the node-opcua client `packages/node-opcua-samples/bin/simple_client.js -e "opc.tcp://SOMEIP:SOMEPORT" -n="SOMENODE"`
3. check the result

**Expected behavior**
Correct authentication and possibility to work with the server.

**Gist**
From Schneider Electric support:
The OPC UA stack used in Schneider Electric products always sends a Null value to the ServerNonce when the SecurityPolice and SecurityMode are configured to "None" allowing Anonymous user token (this won't be the case for security enabled). There are a lot of topics on GitHub about this behavior of the FreeOPCUA client and the reason is that the OPC UA foundation leaves up to the developer to implement or not the ServerNonce with SecurityPolice and SecurityMode are configured to "None" allowing Anonymous user token.
" Depending upon on the SecurityPolicy and the SecurityMode of the SecureChannel, the exchange of ApplicationInstanceCertificates and Nonces may be optional and the signatures may be empty. See OPC 10000-7 for the definition of SecurityPolicies and the handling of these parameters."  - https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.2/

As you can see, currently the most part of the commercialized OPC UA client do not use the ServerNonce with the security settings of your test and some posts on GitHub suggest the developer to remove the ServerNonce from the client.

* my request is related to node-opcua acting as a  OPCUA CLIENT
* I have installed node-opcua as a package ( using npm install )
* I am using an application that uses node-opcua
   - node-red
   - node-opcua-logger: https://github.com/coussej/node-opcua-logger
        
* Device: desktop pc
* OS version: ubuntu 18.04
* Description of the other OPCUA system I am trying to connect to:
     - Name: Schneider Electric PLC
     - Version: M580
     - Manufacturer/Software vendor: Schneider Electric
     

* **node-opcua version:**  :
influx-opcua-logger@2.0.0-alpha /path/to/node-opcua-logger
└── node-opcua@2.1.9

* **Node:** 
   `node --version` = v12.18.0

erossignon, 2020-11-03:
Thank you for reporting this.

We are aiming at UASpecification 1.04 

Beware that a server that does not provide a serverNonce is not secure and will not be able to allow the client to encrypt the password in case of username/password authentication. 

see  https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-19135

Therefore, servers that do not provide ServerNonce can be considered vulnerable...

There is one single case where not providing serverNonce would be valid: when the server  only provide Security.none, and only anonymous connection. I just hope that such servers are rare in the field.

Nethertheless, nothing should prevent NodeOPCUA Client to attempt a connection and should handle serverNonce being null.
However, the client should only accept serverNonce  === null for  SecureChannel security Mode=None and reject it with an error (as it does now) only if the session user identity is not anonymous.

> ... some posts on GitHub suggest ...
can you elaborate?

Also refer to Specification in CreateSessionResponse:
[image]

[image]

erossignon, 2020-11-03:
I have created two extra unit tests to verify the behavior of the client when the server returns a serverNonce === null.
(see https://github.com/node-opcua/node-opcua/commit/dcaa0fa35cbe0e3f81fdd775ddf9e59a4d429426)
The client behavior is as expected:
 - client will accept the connection if serverNonce == null and userIdentity is Anonymous
 - client will reject the connection of serverNonce == null and userIndetity is UserName/password.

rational : The client takes the initiative to drop the connection, in this case, to prevent a clear password to be sent on the network.

So I don't think we should relax the security behavior of the client, to cope with an unsecured server.

Please convince me otherwise if you can.

erossignon, 2020-11-03:
Thinking harder , I suspect that may be Schneider PLC is sending a Nonce of Length = 0 ( serverNonce = [] ) instead of no Nonce at all (serverNonce = null). this may defeat the client.

Can you activate the dump and send a log ?

you will need to make sure that the NODEOPCUADEBUG environment variable is set to TRACE-REQUEST-RESPONSE
````
set NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE
````

Rifasca, 2020-11-04:
Hi,
I am also facing similar issue when I try to run the code. Need help. But when I tried running influx-opcua-logger_v2.0.0-alpha_win-x64.exe, its working fine and I am getting the response.
[2020-11-04T12:06:36.561] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2020-11-04T12:06:36.562] [INFO] main - Initialising influxClient
[2020-11-04T12:06:36.571] [INFO] influx - localhost:8086 responded in 6ms running 1.7.6
[2020-11-04T12:06:36.572] [INFO] main - Initialising buffer
[2020-11-04T12:06:36.574] [INFO] main - Connecting OPCUA
[2020-11-04T12:06:36.604] [FATAL] main - Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Internal Error)
    at Object.start (D:\SPIA\opcua\node-opcua-logger-2.0.0-alpha\src\opcua.js:60:11)
[2020-11-04T12:06:36.605] [INFO] buffer - Persisting membuffer.

mirkocomparetti-synesis, 2020-11-04:
I did clone the repository right now and this is what I get now
`$ node packages/node-opcua-samples/bin/simple_client.js -e "opc.tcp://SOMEIP:SOMEPORT" -n="ns=2;s=SOMEVAL"     securityMode        =  1
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#None
timeout             =  20000
 monitoring node id =  NodeId {
  identifierType: 2,
  value: 'SOMEVAL',
  namespace: 2 }
 connecting to  opc.tcp://SOMEIP:SOMEPORT
    strategy { initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0 }
 Connected ! exact endpoint url is  opc.tcp://SOMEIP:SOMEPORT
endpoint                   Application URI              Product URI                  Application Name     Security Mode  securityPolicyUri                                Type    certificate  discoveryUrls
-------------------------  ---------------------------  ---------------------------  -------------------  -------------  -----------------------------------------------  ------  -----------  -------------------------
opc.tcp://SPEAr1300:SOMEPORT  urn:nxtControl:EcoRT:Server  urn:nxtControl:EcoRT:Server  Ecostruxure Runtime  None           http://opcfoundation.org/UA/SecurityPolicy#None  Server  ...          opc.tcp://SPEAr1300:SOMEPORT

Identify Token for : Security Mode= 1  Policy= http://opcfoundation.org/UA/SecurityPolicy#None
policyId         tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
---------------  ---------  ---------------  -----------------  -----------------
Anonymous_Token  0          null             null               null

Server Certificate :

Options =  1 http://opcfoundation.org/UA/SecurityPolicy#None
 reconnecting to  opc.tcp://SOMEIP:SOMEPORT
 Connected ! exact endpoint url is  opc.tcp://SOMEIP:SOMEPORT
 session created
 sessionId :  ns=0;i=3674259713
 --- NAMESPACE ARRAY ---
 Namespace  0   :  http://opcfoundation.org/UA/
 Namespace  1   :  urn:nxtControl:EcoRT:Server
 Namespace  2   :  http://se.com/EcostruxureSystemExpert
 -----------------------
---------------------------------------------------- All Event Types

 -----------------------
--------------------------------------------------------------- Enumerate all Condition Types exposed by the server
└─ ConditionType
   ├─ DialogConditionType
   └─ AcknowledgeableConditionType
      └─ AlarmConditionType
         ├─ LimitAlarmType
         │  ├─ ExclusiveLimitAlarmType
         │  │  ├─ ExclusiveLevelAlarmType
         │  │  ├─ ExclusiveDeviationAlarmType
         │  │  └─ ExclusiveRateOfChangeAlarmType
         │  └─ NonExclusiveLimitAlarmType
         │     ├─ NonExclusiveLevelAlarmType
         │     ├─ NonExclusiveDeviationAlarmType
         │     └─ NonExclusiveRateOfChangeAlarmType
         ├─ DiscreteAlarmType
         │  └─ OffNormalAlarmType
         │     ├─ SystemOffNormalAlarmType
         │     │  └─ CertificateExpirationAlarmType
         │     ├─ TripAlarmType
         │     ├─ InstrumentDiagnosticAlarmType
         │     └─ SystemDiagnosticAlarmType
         └─ DiscrepancyAlarmType

 -----------------------------------------------------------------------------------------------------------------
 -------------------------------------------------------------- Alarms & Conditions ------------------------
 -----------------------------------------------------------------------------------------------------------------
 ----------------------------------------------------------  Testing QueryFirst
 Server is not supporting queryFirst err=  cannot makeExpandedNodeId out of i=58
 ----------------------------------------------------------  Create Subscription
started subscription : 16842753
 revised parameters
  revised maxKeepAliveCount   12  ( requested  12)
[... 4 lines cut ...]
Monitoring monitor_a_variable_node_value
 Monitoring node  ns=2;s=SOMEVAL
(node:2552) UnhandledPromiseRejectionWarning: Error:  serviceResult = BadServiceUnsupported (0x800b0000)
    at process_request_callback (/home/synesisbnc/test2/node-opcua/packages/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:56:15)
    at ClientSecureChannelLayer._on_message_received (/home/synesisbnc/test2/node-opcua/packages/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:509:9)
    at MessageBuilder.ClientSecureChannelLayer.messageBuilder.on (/home/synesisbnc/test2/node-opcua/packages/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:147:18)
    at MessageBuilder.emit (events.js:198:13)
    at MessageBuilder._decodeMessageBody (/home/synesisbnc/test2/node-opcua/packages/node-opcua-secure-channel/dist/source/message_builder.js:209:26)
    at MessageBuilder._feed_messageChunk (/home/synesisbnc/test2/node-opcua/packages/node-opcua-transport/dist/source/message_builder_base.js:187:18)
    at PacketAssembler.MessageBuilderBase._packetAssembler.on (/home/synesisbnc/test2/node-opcua/packages/node-opcua-transport/dist/source/message_builder_base.js:53:68)
    at PacketAssembler.emit (events.js:198:13)
    at PacketAssembler.feed (/home/synesisbnc/test2/node-opcua/packages/node-opcua-packet-assembler/dist/packet_assembler.js:51:18)
    at MessageBuilder.feed (/home/synesisbnc/test2/node-opcua/packages/node-opcua-transport/dist/source/message_builder_base.js:86:35)
(node:2552) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with . [... cut ...]
ns=2;s=SOMEVAL  value has changed to Variant(Scalar<Null>, value: <null>)
ns=2;s=SOMEVAL  value has changed to Variant(Scalar<Float>, value: 41.99998474121094)`

so now it connects fine without problems, but i did not expect that.

I had the confirmation from Schneider Electric that they are returning FFFF as ServerNonce with anonymous (that is the only supported one by their server so far). See screenshot attached from their support (I also have the wireshark dump, but the file cannot be uploaded here).

Thanks
[image]

erossignon, 2020-11-04:
Available in node-opcua@2.24.0

closed 2020-11-04 by erossignon