node-opcua/node-opcua #906: The more subscriptions active, the more data loss

k1r0s, 2020-11-03:
Number of active subscription dramatically reduces response time per node process.

on the server: 14 variables with 50ms interval change

running one subscription -> no data loss

running 14 subscriptions -> 3/4~ changes are lost

erossignon, 2020-11-03:
This is very uncommon to have one subscription per monitored item, In practice, the client creates very few subscriptions and attaches as many monitored items as possible to the same subscription. 
Please provide the code of your application for review as a gist  ( as suggested in the template form )
I suspect as well that buffers and queue size might not be appropriately set up.
You may need to revisit the architecture of your server/client application. we can provide professional support to help you with this ( contact@sterfive.com).

k1r0s, 2020-11-03:
I was likely using OPCua wrongly coz I was creating a subscription for every variable. Just realized that a single subscription can host many monitored items. 

So the performance is even better. 

Now running my tests I have 1/20 loss. Is it normal or expected?

erossignon, 2020-11-03:
@k1r0s , I would need to see the code that you're using on both ends to spot a potential issue. Contact me privately if you cannot share the code on a public forum.

k1r0s, 2020-11-03:
I may contact you privately but please do answer my question: Its possible to avoid data loss or rather, is possible to get notified of every change in monitored items? 

**My question is whether or not node-opcua (js implementation) is compliant with 100% data integrity with changes happening on 10 millisecond interval basis**

erossignon, 2020-11-07:
By default,  node-opcua servers restrict monitoring sampling intervals to be above 50ms. This is a setting that can be tuned.

Data loss can be avoided by setting proper parameters for queue size.

Also, OPCUA distinguishes two types of variables:

- analog variables that change continuously. in this case, data value harvesting is made using a sampling interval of the monitored item.

- discrete variables that may change at any time for which we do not want to lose any value change. in this case, data harvesting is made using a 'change event base' mechanism, this mode is activated automatically when the variable exhibit aminimumSamplingInternval = 0 and clients specify zero as a sampling rate as well.

Please make sure you are using the right scheme, using a ridiculous low sampling rate is not the right approach for monitoring discrete variables.

k1r0s, 2020-11-07:
Thanks for the detailed answer. I'll soon provide a sample code and needed context.

closed 2022-01-04 by k1r0s