node-opcua/node-opcua #909: Is there API to generate binary namespace file like c/c++ xml2bin?

GeneL, 2020-11-06:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - ( ) support request 

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

* **Typscript:** 
   `tsc --version` =

erossignon, 2020-11-06:
I guess that you are using UAAutomation as a target system.

There is an API to generate NodeSet2.Xml files, but none to convert to bin format.
We have a very efficient XML loader and do not need o convert XML to binary to gain load speed.

You may want to refer to https://documentation.unified-automation.com/uasdkhp/1.1.0/html/address_space_generation.html
to convert XML 2 bin .

erossignon, 2020-12-11:
OPC Foundation is working on specifying a JSON format that should be less verbose than XML.

erossignon, 2021-01-23:
closing due to inactivity

closed 2021-01-23 by erossignon