node-opcua/node-opcua #912: Cannot connect to example simple server using the OPC Foundation OPC Reference Client

afontaine-chipkin, 2020-11-14:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - (x) support request 

**Current behavior**

After following the instructions in the readme to install node-opcua as a npm package, I create an app.js with the example server code found here:  

(http://node-opcua.github.io/) in the Creating a simple Server section. Specifically clicking on the View Code button which takes you to the following repo: (https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/sample_server.js)

When running the application, the primary endpoint url that is printed out is: opc.tcp://SHDWKAIGAN-PC:4334/UA/MyLittleServer

When putting this into the OPC Reference Client, I get the following error:
[image]

Using another OPC Client tool, I get the following error:
Server did not return an EndpointDescription that matched the one used to create the secure channel. = BadSecurityChecksFailed.

*Note*: When I was testing out this package a few months ago, this was working fine.

**Expected behavior**

Both of the clients that I tested the example server with should connect and display the folders and objects of the example server.

**Minimal reproduction of the problem with instructions**

  1. Follow instructions to install node-opcua:
```
$ mkdir mytest
$ cd mytest
$ npm init 
$ npm install node-opcua --unsafe-perms
```
  2. Create app.js and copy the code in from: (https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/sample_server.js)

  3. Run the app.js
```
$ node app.js
```
  4. Confirm that server is running and copy the endpoint url into the OPC Foundation OPC Reference Client tool and connect without using security.  *Note* this error occurs when attempting to connect with or without using security.
  ...

gist:
See Reproduction steps 1 and 2
  
**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
   
    - (x) Windows : version : 10 19041.630
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :  
    
* **node-opcua version:**  :

v2.26.0
also tested on v2.21.0

* **Node:** 
   `node --version` = 12.18.2

afontaine-chipkin, 2020-11-16:
I investigated this more, and found that these issues started with version 2.20.0.

When I test using 2.19.0, everything works.

Searching through the issues I found issue #884 and #881 that describes this and it pointed me to the version 2.20.0 release notes here:  https://github.com/node-opcua/node-opcua/releases/tag/v2.20.0

But even with this, why would the sample server code not work?  Is there something more that we would have to add?

Graos, 2020-11-17:
Endpoint URL exposed by the server should exactly match with the endpoint URL specified in client request. In your case SHDWKAIGAN-PC should be in lower case. Try giving opc.tcp://shdwkaigan-pc:4334/UA/MyLittleServer in the client. I hope it should work.

erossignon, 2020-11-30:
node-opcua@2.28.0 now address the issue of endpoint case sensitivity. @afontaine-chipkin  can you check that issue is now resolved and confirm ?

afontaine-chipkin, 2020-12-01:
Hello @erossignon,

Thank you for the update.  I hope to test this out later this week.  I will let you know if it resolves the issue.

erossignon, 2020-12-11:
[node-opcua@2.28.3](https://github.com/node-opcua/node-opcua/releases/tag/v2.28.3) patch has been released with a better fix for this.
@afontaine-chipkin can you tryit and confirm the issue is resovled now

MarcoSeb, 2021-01-06:
I had the same issue after upgrading node-opcua to 2.28.0.
With 2.28.3 the connection with the "Unified Automation UaExport" works again.
Same with the FreeOpcUa Client....

erossignon, 2021-01-25:
So closing thanks,

closed 2021-01-25 by erossignon