node-opcua/node-opcua #913: OPCUAClient failing to generate client?

adrfantini, 2020-11-18:
Hi!
I am trying to reproduce some old code that still works on some antiquate machine, with an old Node version and `node-opcua` 0.7.4. However, I can't seem to be able to do so.

This is the original code:
```js
const opcua = require("node-opcua");
const certPath = '/my/cert/location/';   // change this for the location of your certificates
const options = {
        endpoint_must_exist: false,
        keepSessionAlive: true,
        reconnectOnFailure: true,
        privateKeyFile: certPath + "/private_key.pem",
        certificateFile: certPath + "/client_selfsigned_cert_1024.pem",
        keepPendingSessionsOnDisconnect: true,
        connectionStrategy: {
            maxRetry: 10,
            initialDelay: 2000,
            maxDelay: 10*1000
        }
}
client = new opcua.OPCUAClient(options);

client.on("someEvent", function(e){
    // do something
});
```
Again, this is working in our old system (which cannot be accessed for tests unfortunately).
In a newer system with Node 12 and `node-opcua` 2.22.0 (or above, I tested different versions), the `client` object has no `on` method, thus of course the code fails.

I tested with a combination of all the options above, including an empty `options` field. No matter the `options` that are passed, I was never able to obtain a `client.on` function. In fact, it appears that all the methods listed in the [documentation](https://node-opcua.github.io/api_doc/2.0.0/classes/opcuaclient.html) are missing, as if the client's creation is failing silently.

For example:
```js
const opcua = require("node-opcua")
client = new opcua.OPCUAClient({})
client
//OPCUAClient {}
client.constructor
// [Function: OPCUAClient]
client.on
// undefined
```

What am I missing?

erossignon, 2020-11-20:
node-opcua 0.7.4 is rather old, a lot of changes have been introduced since. and a major API break took place in 2.0.0
Please check the wiki for breaking or [node-opcua by example book](https://leanpub.com/node-opcuabyexample)

adrfantini, 2020-11-23:
> node-opcua 0.7.4 is rather old, a lot of changes have been introduced since. and a major API break took place in 2.0.0
> Please check the wiki for breaking or [node-opcua by example book](https://leanpub.com/node-opcuabyexample)

My bad for not checking the wiki. Thanks!

closed 2020-11-23 by adrfantini