node-opcua/node-opcua #914: Crash when the nodeset CommercialKitchenEquipment is not found (lacks '.xml' at code)

IagoSRL, 2020-11-18:
**Current behavior**
When requiring the module `node-opcua`, a crash happens because cannot find an xml file that comes bundle with it. Error message:
`Error: Please make sure that nodeset can be found in node_modules\node-opcua-nodesets\nodesets` (I removed the absolute part of the path).

**Describe the bug**
I checked and the folder exists, so debugged the nodesets and it founds lot of files in the folder, except for `Opc.Ua.CommercialKitchenEquipment.NodeSet2`, the error is thrown at the line trying lo locate that file.

The JavaScript file is at the `node-opcua-nodesets` module, exactly [at this line in the source code](https://github.com/node-opcua/node-opcua/blob/ede53ccd8f4c76bcbf3997c3237cb0a35aca59ae/packages/node-opcua-nodesets/source/index.ts#L47)

The source code line is:
`const commercialKitchenEquipmentNodeSetFilename = constructNodesetFilename("Opc.Ua.CommercialKitchenEquipment.NodeSet2");`

**It lacks the ".xml" extension**, found in the other lines.

**To Reproduce**
- `npm i node-opcua`
- `node`
- `require('node-opcua');`  (Inside the node REPL)

**Proposed fix:**
It seems could get fixed by adding the XML extension at the filename at the commented source code line, but it's still strange to me why sometimes throws and sometimes not.

**QUESTION:** Why I was able to use the module without getting this error at some use cases, but in others so simple as requiring the module it throws this error?

**ANSWER:** *(I researched a bit more when writting this issue and found, partially, the answer by myself :-)* There is a condition at the code that only throws error when there is no argument given to the execution of Nodejs, basically meaning that only throws when requiring the module from the Nodejs REPL. At any other case (like `node my-app.js`), won't throw an error, it will just try to build the path to the forgotten file in a different way without throwing even if the constructed path doesn't exist.

To be honest, I found this whole behavior confusing (how is locating paths and managing the error/not error), but cannot say if is a thing for another issue or there is a good reason why this is the correct way and expected behavior by anyone else. Please don't take me wrong, I am not questioning the design decission around this, I just wanted to point out what I found just in case is a bug.

**Expected behavior**
Being able to load `node-opcua` from the node REPL.

 * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: Windows 10
    
    - (x) Windows : version : Windows 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  : 2.26.0

* **Node:** 
   `node --version` = 12.18.4

erossignon, 2020-11-20:
Yes, I spotted too, should be fixed in next version.

IagoSRL, 2020-11-20:
Thanks @erossignon.

erossignon, 2020-11-21:
fixed in node-opcua@2.27.0

closed 2020-11-23 by erossignon