node-opcua/node-opcua #918: Exception in UATwoStateDiscrete._post_initialize function while loading the attached nodeset file.

Graos, 2020-11-24:
**Current behavior**
Exception in UATwoStateDiscrete._post_initialize while loading the attached nodeset file.

**Describe the bug**
Exception in UATwoStateDiscrete._post_initialize while loading the attached nodeset file.

**To Reproduce**
Steps to reproduce the behavior:
1. load the attached nodeset file to the server.
2. ......
3.......

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
24.11.2020, 18:06:59[Info] --> NODE-OPCUA stack version : 2.27.1
24.11.2020, 18:06:59[Info] --> NODEJS version : v14.15.1

(node:24072) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'on' of undefined
    at UATwoStateDiscrete._post_initialize (\node_modules\node-opcua-address-space\dist\src\data_access\ua_two_state_discrete.js:20:25)
    at promoteToTwoStateDiscrete (\node_modules\node-opcua-address-space\dist\src\data_access\ua_two_state_discrete.js:71:11)
    at \node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:26:25
    at Generator.next (<anonymous>)
    at \node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:4:12)
    at promoteObjectAndVariablesInNamespace (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:19:12)
    at Object.<anonymous> (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:41:13)
    at Generator.next (<anonymous>)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:24072) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:24072) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:24072) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'on' of undefined
    at UATwoStateDiscrete._post_initialize (\node_modules\node-opcua-address-space\dist\src\data_access\ua_two_state_discrete.js:20:25)
    at promoteToTwoStateDiscrete (\node_modules\node-opcua-address-space\dist\src\data_access\ua_two_state_discrete.js:71:11)
    at \node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:26:25
    at Generator.next (<anonymous>)
    at \node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:4:12)
    at promoteObjectAndVariablesInNamespace (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:19:12)
    at Object.<anonymous> (\node_modules\node-opcua-address-space\dist\source\loader\namespace_post_step.js:41:13)
    at Generator.next (<anonymous>)
(node:24072) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
   

  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  : 2.27.1

* **Node:** 
   `node --version` = 14.15.1
[NCC.zip](https://github.com/node-opcua/node-opcua/files/5589912/NCC.zip)

erossignon, 2020-11-26:
In fact, the NCC.xml contains errors, the TwoStateDiscrete Variable must have a  TrueState and FalseState Property. There are missing on 
see https://reference.opcfoundation.org/v104/Core/VariableTypes/TwoStateDiscreteType

for instance 
```xml line63683
   <UAVariable DataType="Boolean" NodeId="ns=2;s={NCP-NC-TWO-STATE-DISCRETE-SIGNAL-TYPE-OVERRIDE-VALUE}" BrowseName="2:OverrideValue" AccessLevel="3">
        <DisplayName>OverrideValue</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=2373</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=2;s={NCP-NC-TWO-STATE-DISCRETE-SIGNAL-TYPE}</Reference>
        </References>
        <Value>
            <uax:Boolean>false</uax:Boolean>
        </Value>
    </UAVariable>
```

should have those 2 properties

Graos, 2020-11-27:
Thanks. My mistake.

Graos, 2020-11-27:
Not an issue with the server.

closed 2020-11-27 by Graos