node-opcua/node-opcua #919: Newer library has issue, clients cannot connect anymore

kallegit1950, 2020-11-25:
In the past I was able to run the sample server with the UA expert client 1.4.1 and 1.5.1. Now with the latest library (not sure what version broke it) I cannot connect any more to the opc server. To reproduce run the opc sample server and try to connect once with UA expert 1.4.1 and once with 1.5.1, the latter will fail with that error message

19:11:44.607 | Server Node        | localServerOPCUA_test          | Endpoint: 'opc.tcp://secret:26543'
19:11:44.607 | Server Node        | localServerOPCUA_test          | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
19:11:44.608 | Server Node        | localServerOPCUA_test          | ApplicationUri: 'urn:NodeOPCUA-Server-default'
19:11:44.608 | Server Node        | localServerOPCUA_test          | Used UserTokenType: Anonymous
19:11:44.758 | Server Node        | localServerOPCUA_test          | Error 'BadConfigurationError' was returned during ActivateSession
19:11:44.769 | Server Node        | localServerOPCUA_test          | Connection status of server 'localServerOPCUA_test' changed to 'Disconnected'.

running the latest node 14.15.1 on windows 10.

UA Expert I consider as standard tool, if not known; https://www.unified-automation.com/de/downloads/opc-ua-clients.html

Any idea what this might be or what I can do to provide more information?

Graos, 2020-11-26:
might be same as **OPC UA client (UAExpert) is not connecting to Node-OPCUA server anymore (from version 2.20.0) #884** . Endpoint URL exposed by the server should exactly match with the one provided in the client side. It is even case sensitive. Please check.

kallegit1950, 2020-11-26:
Sounds exactly like this issue, but how do I find out the exact endpoint I shall use? When I run the sample server I see this as output

Server is now listening ... ( press CTRL+C to stop)
opc.tcp://SoMe.Server:26543 1 http://opcfoundation.org/UA/SecurityPolicy#None
opc.tcp://SoMe.Server:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
opc.tcp://SoMe.Server:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic256
opc.tcp://SoMe.Server:26543 2 http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
opc.tcp://SoMe.Server:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
opc.tcp://SoMe.Server:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic256
opc.tcp://SoMe.Server:26543 3 http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256

So in UA Expert I copied the string opc.tcp://SoMe.Server:26543 and used it in there. When you run the sample server what do you enter as address in UA expert?

When running the test script attached to the other issue I get

>node getEndpointBasic.js opc.tcp://SoMe.Server:26543
connecting to  opc.tcp://SoMe.Server:26543
still connecting  opc.tcp://SoMe.Server:26543

the last line repeats forever. I have stopped it after a few minutes.

I wonder if I can enhance the server code that it tells me what endpoint I have to use.

Graos, 2020-11-26:
since you have mentioned that you use Anonymouse user token policy i expect the URL **opc.tcp://SoMe.Server:26543** to work when provided in the UAexpert.

kallegit1950, 2020-11-26:
That is what I would think as well, but it doesn't. Is there any idea, maybe some other test client, we can use to debug that issue? Does the UA expert 1.5.1 work for someone else with the test server provided on github? If it works for someone else it is maybe an OS configuration issue locally, if it doesn't work for anyone then it is general issue. Any ideas are welcome. Thanks for your input.

Graos, 2020-11-26:
I am using UaExpert 1.5.1 . It works seamlessly with node-opcua server. Please share the server sample that you use. I am sure there should be some mismatch in that.

kallegit1950, 2020-11-26:
Hi, I am using the official sample server from here https://raw.githubusercontent.com/node-opcua/node-opcua-sampleserver/master/server.js the opc-ua libraries are 2.27.1.

Graos, 2020-11-26:
strange i use the same UaExpert version and Node-opcua version . I don't see any issue. see below the connection snapshot
[image]

kallegit1950, 2020-11-26:
okay, I run it now on a different PC and there it works as well. So I assume my windows image I used before does something else with name resolution or something like that. Unfortunately I have to use that image. Any option to debug what is the difference?

Graos, 2020-11-26:
Best option is to use network capture tool like Wireshark to check the packet been sent from your client to the server.

mikakaraila, 2020-11-26:
Etienne has been fixing this... "known" issue. New version should be available soon.

kallegit1950, 2020-11-26:
thanks for the hint. So I need to wait until that fix is released? Is there any easy way that I can apply these changes manually to my server to get working again until a new release is out?

mikakaraila, 2020-11-26:
Just wait, I expect Etienne will get it ready today/tomorrow.

erossignon, 2020-11-30:
node-opcua@2.28.0 has been released today and shall fix the issue,  @kallegit1950 please confirm

kallegit1950, 2020-12-01:
@erossignon unfortunately  exactly the same behavior.

mikakaraila, 2020-12-01:
In my case I forgot to update node-opcua-pki and node-opcua-crypto. Check those first!

mikakaraila, 2020-12-01:
I studied code bit more and I expect matchUri don´t get actual endpoints. It is filtered away before comparison.

const epFiltered = endpointUrl ? ep.filter((e) => matchUri(e.endpointUrl, endpointUrl)) : ep;

mikakaraila, 2020-12-01:
Or some clients are expecting resourcePath is always after port number.

resourcePath: "/"  // server options

kallegit1950, 2020-12-01:
I started from scratch the server plus a package.json file refering only to the opc-ua server code the rest was pulled automatically. So I believe pki and crypto are on the latest release, they are on 2.2.2 and 1.4.4 as far as I can see on my local installation. I have checked the respective package.json files in the node_modules folder.

For the resourcePath I tried all kind of combinations, of specifying it the code but not in the connection and vice versa. Tried different paths all no luck.

Since UA Expert 1.5.1 works for you but not for me on my specific corporate image it must be something which is very specific to my image. Is there somewhere an opc-ua client which can deliver some more debug output I could try to use to figure out the issue I have on my local installation?

kallegit1950, 2020-12-01:
I downloaded the opc-ua test client https://github.com/node-opcua/node-opcua-sample and tried to use this to connect. Of course I have changed line 9

`const endpointUrl = "opc.tcp://SERVER.domain.com:26543";`

to point to my server. Client and server run on the same laptop. I am getting this:

`>node simple_client_es8.js
retrying connection
retrying connection
Error !!! Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (getaddrinfo ENOTFOUND server.domain.com)`

What puzzles me is that it wants to use some client certificate to connect, I cannot see where the client or the server mandates a cert, the server even says it is fine if a client connects without a cert. The other thing the server outputs that is waits for connection on SERVER.domain.com, but the client says it needs a cert for server.domain.com.

Does that test lead to a wrong direction? If yes ignore it, but maybe it helps.

mikakaraila, 2020-12-02:
Missing this from the code?

await certificateManager.initialize()

kallegit1950, 2020-12-02:
@mikakaraila I am afraid I don't understand your question  or hint. Your code snippet I am not able to see in the server nor the client code I am using. https://github.com/node-opcua/node-opcua-sample/blob/master/simple_client_es8.js  and https://raw.githubusercontent.com/node-opcua/node-opcua-sampleserver/master/server.js Are you suggesting that this is missing somewhere? If yes where and why is that not an issue for others? I am right now not concerned about security, no need for certificates as long as the standard code don't require them. Can be as insecure as it wants (for now) as long as I am able to connect.

mikakaraila, 2020-12-02:
In the sample server there is no certificate manager... you should add it if you want to use certificates.

kallegit1950, 2020-12-02:
I do not want to use certificates. In UA expert I have set

Security Policy: None
message Security Mode: None
Authentication Settings: Anonymous

The test with node client was only if it give an additional hint. At the end I don't care about the node client. My target is to use UA expert 1.5.1 to connect to the server https://raw.githubusercontent.com/node-opcua/node-opcua-sampleserver/master/server.js with no security, as simple as it gets.

mikakaraila, 2020-12-02:
It can be some old dependencies, try to run: "npm dedupe"
It has cleaned my problems couple of times.

I am using 2.28.0 without any problems. So it must be something on your side.

kallegit1950, 2020-12-02:
I agree that the issue on my side. But, my assumption is that others run into similar issues and I am not the only one with a client configuration like mine. The question is what can I do to figure out what happens on my end that I can change (or adapt the server that is works again).

npm dedupe didn't help.

erossignon, 2020-12-11:
[node-opcua@2.28.3](https://github.com/node-opcua/node-opcua/releases/tag/v2.28.3) patch has been released with a fix for this.

@kallegit1950 please confirm

kallegit1950, 2020-12-11:
@erossignon still the same issue. If there is any debug tool I can use to get to the root cause then please let me know.

erossignon, 2020-12-12:
@kallegit1950  in this case, contact me for professional support.

kallegit1950, 2021-01-13:
Tested again with 2.29.0, that version is working fine with UA Expert 1.5.1. So I assume some update between 2.28.3 and 2.29.0 has made the difference. From my point of view this bug report can be closed. Thanks very much for your patience and all the help!

closed 2021-01-14 by erossignon