node-opcua/node-opcua #920: pkg + node-opcua server

Mikuz, 2020-11-26:
**Current behavior**

This problem is very similar to issue #530, but it's solution is not working.
I'm want to build self executable binary sample_server.js on windows with pkg tool (https://www.npmjs.com/package/pkg)
But what I get:

**######### ERROR #########**

Error    TypeError: Cannot read property 'objects' of null
  - server_engine.js:777 prepareServerDiagnostics
    [Node]/[node-opcua-server]/dist/server_engine.js:777:47

  - server_engine.js:823
    [Node]/[node-opcua-server]/dist/server_engine.js:823:13

  - load_nodeset2.js:1297
    [Node]/[node-opcua-address-space]/dist/source/loader/load_nodeset2.js:1297:25

  - task_queues.js:97 processTicksAndRejections
    internal/process/task_queues.js:97:5

  - task_queues.js:66 process.runNextTicks [as _tickCallback]
    internal/process/task_queues.js:66:3

  - bootstrap.js:1376 Function.Module.runMain
    pkg/prelude/bootstrap.js:1376:13

  - run_main_module.js:17
    internal/main/run_main_module.js:17:47

(node:10924) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'objects' of null
    at prepareServerDiagnostics (C:\snapshot\Node\node_modules\node-opcua-server\dist\server_engine.js:777:47)
    at C:\snapshot\Node\node_modules\node-opcua-server\dist\server_engine.js:823:13
    at C:\snapshot\Node\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:1300:13
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at process.runNextTicks [as _tickCallback] (internal/process/task_queues.js:66:3)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1376:13)
    at internal/main/run_main_module.js:17:47
(Use `OPC-Server-1.0.1-x64 --trace-warnings ...` to show where the warning was created)
(node:10924) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:10924) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
session: cannot read Server_NamespaceArray
Error    TypeError: Cannot read property 'objects' of null

  - server_engine.js:777 prepareServerDiagnostics
    [Node]/[node-opcua-server]/dist/server_engine.js:777:47

  - server_engine.js:823
    [Node]/[node-opcua-server]/dist/server_engine.js:823:13

  - load_nodeset2.js:1297
    [Node]/[node-opcua-address-space]/dist/source/loader/load_nodeset2.js:1297:25

(node:10924) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'objects' of null
    at prepareServerDiagnostics (C:\snapshot\Node\node_modules\node-opcua-server\dist\server_engine.js:777:47)
    at C:\snapshot\Node\node_modules\node-opcua-server\dist\server_engine.js:823:13
    at C:\snapshot\Node\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:1300:13
(node:10924) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

**######### END OF ERROR #########**

**Describe the bug**
Cannot make working single exe file from example server.js

**To Reproduce**
Steps to reproduce the behavior:
1. Download node-opcua
2. Create app.js, cloned from sample_server.js
3. Insert     "pkg": {
    "scripts": [], 
    "assets": [ 
      "/node_modules/node-opcua-server/certificates/server_selfsigned_cert_2048.pem",
	  "/node_modules/node-opcua-server/certificates/PKI/own/private/private_key.pem"
    ]
  }, into package.json
4. Use PKG, `pkg .  --target host --output myapp-1.0.0-x64.exe` to get exe, 
5. Use exe and get error.

**Expected behavior**
Working correctly OPC UA server.

**Gist**
gist:
  * (X)  my request is related to node-opcua acting as a OPCUA SERVER 
  * (X) I have installed node-opcua as a package ( using npm install )
  * Device: PC
  * OS version: Windows 10
    - (X) Windows : version : 10  

* **node-opcua version:**  : 2.27.1

* **Node:** 
   `node --version` = v14.15.1

**Other notes**
sample_client.js is working flawlessly as standalone exe.

mikakaraila, 2020-11-26:
How you have introduced nodeset XML file in the code?

I am using this:
const standard_nodeset_file = path.join(path.dirname(process.execPath), "nodesets/Opc.Ua.NodeSet2.xml");

Mikuz, 2020-11-27:
Thank You, It works now!
I added Xml to assets:
    "assets": [ 
      "/node_modules/node-opcua-server/certificates/server_selfsigned_cert_2048.pem",
	  "/node_modules/node-opcua-server/certificates/PKI/own/private/private_key.pem",
	  "/node_modules/node-opcua-nodesets/nodesets/Opc.Ua.NodeSet2.xml"
    ]

closed 2020-11-27 by Mikuz