node-opcua/node-opcua #921: Assertion Error on OPCUAServer for X509IdentityToken connections

hellivan, 2020-11-27:
**Current behavior**
OPCUAServer throws an assertion error if a client tries to connect with a X509IdentityToken.

**Describe the bug**
OPCUAServer internally uses module `node-opcua-crypto` which throws an assertion error in `verifyMessageChunkSignature` function if the client sends a PEM formatted certificate.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the OPCUA-Sample-Server with a configured userCertificateManager
2. Try to connect with the OPCUA-Sample-Client that was configured with a `UserIdentityInfo` of type `UserTokenType.Certificate`. `certificateData` should be a buffer, containing a PEM formatted certificate.

**Expected behavior**
The OPCUAServer should not throw the assertion error but handle the X509IdentityToken correctly.

**Stacktrace**
```
The expression evaluated to a falsy value:

  assert(typeof options.publicKey === "string")

AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:

  assert(typeof options.publicKey === "string")

    at Object.verifyMessageChunkSignature (.../opc-ua/node_modules/node-opcua-crypto/dist/source/crypto_utils.js:131:5)
    at Object.RSAPKCS1OAEPSHA256_Verify [as asymmetricVerify] (.../opc-ua/node_modules/node-opcua-secure-channel/dist/source/security_policy.js:175:32)
    at Object.verifySignature (.../opc-ua/node_modules/node-opcua-secure-channel/dist/source/security_policy.js:348:26)
    at OPCUAServer.isValidX509IdentityToken (.../opc-ua/node_modules/node-opcua-server/dist/opcua_server.js:810:42)
    at OPCUAServer.isValidUserIdentityToken (.../opc-ua/node_modules/node-opcua-server/dist/opcua_server.js:897:25)
    at OPCUAServer._on_ActivateSessionRequest (.../opc-ua/node_modules/node-opcua-server/dist/opcua_server.js:1280:16)
    at OPCUAServer.on_request (.../opc-ua/node_modules/node-opcua-server/dist/base_server.js:206:25)
    at OPCUAServerEndPoint.<anonymous> (.../opc-ua/node_modules/node-opcua-server/dist/opcua_server.js:496:26)
    at OPCUAServerEndPoint.emit (events.js:315:20)
    at ServerSecureChannelLayer.<anonymous> (.../opc-ua/node_modules/node-opcua-server/dist/server_end_point.js:495:22)
```

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
 * (*)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (*) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
            

* **node-opcua version:**  : 2.27.1

* **Node:** 
   `node --version` = 14.15.1

hellivan, 2020-11-27:
A fix for this issue was implemented in https://github.com/node-opcua/node-opcua-crypto/pull/7

erossignon, 2020-11-29:
certificate must be either a DER ( in a buffer) or a PEM ( in a string)
Please use the `readCerticate` method to make sure to get the certificate in the right format.

hellivan, 2020-12-06:
Hi @erossignon . Thank you for your fast reply.

Unfortunately my current usecase does not allow me to use the `readCertificate` method which reads the certificate data from a file, since I get my PEM formatted certificate string from a JSON object field.

According to the current type definitions, `certificateData` can only be a `ByteString` which in fact is a type alias for `Buffer`. Therefore I instantiated my `UserIdentityInfo` just like this:
```
{
    type: UserTokenType.Certificate,
    certificateData: Buffer.from(jsonData.certificatePEMstring),
    privateKey: jsonData.privateKeyPEMstring
}
```
So my question now is:
Can I pass my PEM formatted certificate data just like shown in this example or should I use the `convertPEMtoDER` method from `node-opcua-crypto` in order to convert it into a DER Buffer/ByteString first?

erossignon, 2020-12-06:
Yes, try to convert the PEM ascii string to a binary DER using convertPEMtoDER in your case.

hellivan, 2020-12-06:
@erossignon is this required in general or only in this special case?

erossignon, 2020-12-07:
This is the way X509IdentityToken is designed to work:  certificateData must hold a DER certificate in a Buffer.

erossignon, 2021-01-19:
@hellivan is the issue resolved now ? can we close it ?

hellivan, 2021-01-29:
@erossignon issues is resolved.

closed 2021-01-29 by erossignon