node-opcua/node-opcua #922: Errors with self-signed certificates

jadamroth, 2020-12-08:
**Current behavior**

I have an Electron + vue.js app where I intend to use this library as an opcua client, but I am getting a lot of errors with the self signed certificates, which seem to be potentially related (though I intend to make anonymous requests with no encryption requirements from the opcua server).

**Describe the bug**

I installed this library with npm install. When I tried building my electron app, I had to add a node-loader in my webpack configuration because I was getting this build error
```
./node_modules/node-opcua-pki/node_modules/fsevents/fsevents.node

Module parse failed: Unexpected character '�' (1:0)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file.
```

After I added the loader in my webpack rules, my program builds. However, when I try to create my opcua client, 
```javascript
const { OPCUAClient, MessageSecurityMode, SecurityPolicy } = require("node-opcua");
const options = {
        applicationName: "ClientName",
        connectionStrategy: {
            initialDelay: 1000,
            maxRetry: 1
        },
        securityMode: MessageSecurityMode.None,
        securityPolicy: SecurityPolicy.None,
        endpoint_must_exist: false,
    };
    const client = OPCUAClient.create(options);
```

I get this error, which makes sense because the certificates don't copy over into my dist package.
```
cannot locate certificate file /path/to/my/app/node_modules/electron/dist/Electron.app/Contents/Resources/certificates/client_selfsigned_cert_2048.pem
```

So, I manually move the certificates to this directory, which fixes that error, but I get a new error that pops up
```
TypeError: node_opcua_crypto_1.readKeyPem is not a function
```

I see that this path exists in my proper distribution folder, and the function readKeyPem() in read.js. Here's where the error happens in opcua_secure_object.js
```javascript
const node_opcua_assert_1 = __webpack_require__(/*! node-opcua-assert */ "./node_modules/node-opcua-assert/dist/index.js");
const node_opcua_crypto_1 = __webpack_require__(/*! node-opcua-crypto */ "./node_modules/node-opcua-crypto/index_web.js");
function _load_certificate(certificateFilename) {
    const der = node_opcua_crypto_1.readCertificate(certificateFilename);
    return der;
}
function _load_private_key_pem(privateKeyFilename) {
    return node_opcua_crypto_1.readKeyPem(privateKeyFilename);
}
```

**Expected behavior**
To be able to create an opcua client.

- (x) my request is related to node-opcua acting as an OPCUA CLIENT

- (x) I have installed node-opcua as a package ( using npm install )

  - Device: 
    - Macbook pro
  - OS version: 
    - macOS Big Sur version 11.0.1

  - Description of the other OPCUA system I am trying to connect to:
    - A custom opcua server (not from this library) which I have connected to using both Java and Golang opcua client libraries

- **node-opcua version**:  ^2.28.2 but I also tried ^2.27.1

- **Node:**
  `node --version` = Node v12.16.3, Electron 10.1.6

jadamroth, 2020-12-08:
I should also mention that I did review issue #681 which did have some similarities, but it did not fix the issue(s)

Vishal1419, 2020-12-09:
I have seen that you have posted a job on upwork to solve this issue.
So, I believe you are willing to pay, if someone solves your issue.

So, you can pay to the professional support and get your issue solved:
As mentioned in the docs, you can get professional support here: https://www.sterfive.com/

Note: I am not 100% sure that they will fix your issue. Also, I do not have any personal or professional relation with them. I am just advising you.

jadamroth, 2020-12-09:
Thank you for the advice. Simultaneously, I am still trying to solve this issue myself.

I'm confident I am able to program everything required for my application once I have the library properly building with Electron. Just need to get past this first hurdle

mikakaraila, 2020-12-09:
We have build node-opcua Electron app so keep trying. Most probably you should load certificate outside of application, not as build in/included inside application.

jadamroth, 2020-12-09:
Thank you! I will give it a shot

jadamroth, 2020-12-09:
@mikakaraila I still get the same error of -> readKeyPem is not a function()

Out of curiosity, which version of electron and node-opcua are you using? 

Also, what is the build process for your electron app? For example, I'm packaging using vue-cli and webpack to bundle everything needed

motasimali, 2020-12-10:
I have left a message on your job. Leave a reply there. I will help you fix this.

jadamroth, 2020-12-10:
I started modifying the source code in the node_modules which were downloaded from npm install.

I printed out the required object to the console from require("node-opcua-crypto"); and noticed that nothing in dist/source_nodejs was included. 

I resolved the error once I changed node-opcua-crypto/index_web.js from
```javascript
module.exports = require("./dist/source");
```
to match node-opcua-crypto/index.js 
```javascript
module.exports = {
    ...require("./dist/source"),
    ...require("./dist/source_nodejs")
};
```

closed 2020-12-10 by jadamroth

EmanueleCoppola, 2022-07-01:
@jadamroth have you been able to fix this issue (the one that says that the function "readCertificate" is not defined)?
I'm going through a similar error and I haven't been able to fix it yet.

Thanks in advance

jadamroth, 2022-07-01:
I had a lot of issues originally, but it ultimately went away without me having to modify every time. Not really sure why. 

But when I had to fix the error manually, I have these written down for different errors in my readme. Hope this helps

```
# cannot locate certificate file
cd node_modules/node-opcua-client/
cp -R certificates ../../node_modules/electron/dist/Electron.app/Contents/Resources/
```
```
# node_opcua_crypto_1.readKeyPem is not a function
vi node_modules/node-opcua-crypto/index_web.js
```
```
# replace this with below -> module.exports = require("./dist/source");
module.exports = {
    ...require("./dist/source"),
    ...require("./dist/source_nodejs")
};
```

EmanueleCoppola, 2022-07-03:
I'm using Vite + Vue to build this app and I actually fixed it just by setting the Vite build.target configuration to 'es2021'.

I hope this might help someone in the future