node-opcua/node-opcua #925: Problem in certificate thumbprint calculation

Filipecordeiro, 2020-12-28:
**Current behavior**

Using a server with the node-opcua implementation and a client using other implementation (UaExpert): when using a certificate signed by a CA for the server (i.e. not a self-signed certificate), the server throws an error while checking the receiverCertificateThumbprint sent by the client.

**Describe the bug**

The _check_receiverCertificateThumbprint method (node-opcua-secure-channel\source\server\server_secure_channel_layer.ts) is using the whole certificate chain for calculating the certificate thumbprint. While the [node-opcua specification](https://reference.opcfoundation.org/v104/Core/docs/Part6/6.7.2/) is not clear, it does mention the receiverCertificateThumbprint indicates the public key used to encrypt the MessageChunk, and every method that I've tested for calculating certificate thumbprints (including openssl) seems to use just the certificate itself and not the full certificate chain.

I've also found this [issue](https://github.com/OPCFoundation/UA-Java-Legacy/issues/145) in a Java implementation of node-opcua that seems exactly the same problem where they changed it to use the certificate only.

I've changed the code in _check_receiverCertificateThumbprint from `const serverCertificateChain = this.getCertificateChain();` to `const serverCertificateChain = this.getCertificate();` and it solved the problem.

Also had to make the same change in code in the **verifyClientSignature** method (node-opcua-server\source\opcua_server.ts) that seems to **suffer from the same issue (full certificate chain is being used)**.

**To Reproduce**

Steps to reproduce the behavior:

1. Start node-opcua acting as an OPCUA SERVER, using a certificate signed by a CA, providing at least MessageSecurityMode = Sign&Encrypt, and automaticallyAcceptUnknownCertificate = true
2. Try connecting with a client (other than a node-opcua implementation), using MessageSecurityMode = Sign&Encrypt
3. Server throws an error and client disconnects

**Expected behavior**

The client should connect successfully, i.e., the receiverCertificateThumbprint sent by the client should match the certificate thumbprint computed by the server.

**Details**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua from source ( using git clone)
  - (x) I have installed node-opcua as a package ( using npm install )
  - (x) I am using an application that uses node-opcua

    - ( ) node-red
    - (x) other : UaExpert

  - Device: Laptop
  - OS version: Windows 10

    - (x) Windows : version : 10.0.18363
    - ( ) Linux : version : \***\*\_\*\***
    - ( ) MacOs : version : \***\*\_\*\***
    - ( ) Raspbian: version : \***\*\_\*\***
    - ( ) Other : specify :

  - Description of the other OPCUA system I am trying to connect to:

    - Name: UaExpert
    - Version: 1.5.1.331
    - Manufacturer/Software vendor: Unified Automation GmbH
    - link : https://www.unified-automation.com/

- **node-opcua version:** :
  2.16.0

- **Node:**
  10.19.0

erossignon, 2020-12-29:
Using certificate provided by a CA should work.
Please provide source code of the server for investigation. (contact me privately if you cannot share the code in github)

mikakaraila, 2020-12-29:
Did you read issue? 
It is node-opcua server and UaExpert as client and node-opcua server calculates thumbprint from the chain not the certificate itself only.

erossignon, 2020-12-29:
I confirm the issue

erossignon, 2021-01-13:
This should now be fixed in version 2.29 , @Filipecordeiro  can you confirm ?

Filipecordeiro, 2021-03-18:
> This should now be fixed in version 2.29 , @Filipecordeiro can you confirm ?

I can confirm that the issue is fixed. Thank you @erossignon
I apologize for the late reply but I wasn't able to test this until very recently.

closed 2021-03-18 by Filipecordeiro