node-opcua/node-opcua #930: V2.29.0 [NODE-OPCUA-E010] Configuration error : the certificate and the private key do not match !

srogers2112, 2021-01-13:
**Current behavior**

After upgrading the node-opcua package via npm update (from 2.14.0 to 2.29.0), and starting a OPCUAServer, the server fails to start with the following:
```
20:21:33.398Z :verify                        :28    [NODE-OPCUA-E010] Configuration error : the certificate and the private key do not match !
20:21:33.398Z :verify                        :29                      please check the configuration of the OPCUA Server
20:21:33.399Z :verify                        :30                      privateKey=  /Users/srogers/Library/Preferences/NodeOPCUA-Default-nodejs/certificates/own/private/private_key.pem
20:21:33.399Z :verify                        :31     cerficateManager.privateKey=  /Users/srogers/Library/Preferences/NodeOPCUA-Default-nodejs/certificates/own/private/private_key.pem
20:21:33.399Z :verify                        :32                      certificateFile=  /Users/srogers/Library/Preferences/NodeOPCUA-Default-nodejs/certificates/own/certs/certificate.pem
initialized
/Users/srogers/Documents/envieta/MHS/projects/remote_viz/plc-server/test/mock-plc.js:66
        const addressSpace = server.engine.addressSpace;
                                           ^

TypeError: Cannot read property 'addressSpace' of undefined
    at construct_my_address_space (/Users/srogers/Documents/envieta/MHS/projects/remote_viz/plc-server/test/mock-plc.js:66:44)
    at post_initialize (/Users/srogers/Documents/envieta/MHS/projects/remote_viz/plc-server/test/mock-plc.js:111:5)
    at /Users/srogers/Documents/envieta/MHS/projects/remote_viz/plc-server/node_modules/node-opcua-server/dist/opcua_server.js:622:13
[nodemon] app crashed - waiting for file changes before starting...
```

**Describe the bug**
It seems as if something is broken in the certificate and key matches.
It should be noted, that I am not specifying any security requirements or certificates when running server

**To Reproduce**
Steps to reproduce the behavior:

Attempt to start the Server via node start (the .js file containing the server code)
ie.  node ./src/server.js

**Expected behavior**

Expected to work as previously, unless breaking change noted in release notes. Note, I verified the script works in v2.28.3, but changing back to 2.29.0 reproduces the error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

please provide a full  script or program that can help reproduce the issue.
```
/*global require,setInterval,console */
const opcua = require("node-opcua");

// Let's create an instance of OPCUAServer
const server = new opcua.OPCUAServer({
    port: 3030,
    resourcePath: "/SchneiderElectric/Uaserver",
    buildInfo: {
        productName: "MockPlcServer",
        buildNumber: "1",
        buildDate: new Date(),
    },
});

function post_initialize() {
    console.log("initialized");

    function construct_my_address_space(server) {
        const addressSpace = server.engine.addressSpace;

        // Hack to get namespace 2.  See https://github.com/node-opcua/node-opcua/issues/634
        const urn = "urn:mock-pl2:ns2";
        const namespace = addressSpace.registerNamespace(urn);
        const nsId = addressSpace.getNamespaceIndex(urn);

        const device = namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "MockPLC1",
        });

        // Add a bunch of nodes here
            // namespace.addVariable({
            //     componentOf: device,
            //     nodeId: nodeId,
            //     browseName: "foobar",
            //     dataType: dataType,
            //     value: value,
            // });
    }

    construct_my_address_space(server);

    server.start(function () {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl);
    });
}

server.initialize(post_initialize);
```
Put code in a file named server.js and start with node server.js.

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua from source ( using git clone)
  - (x) I have installed node-opcua as a package ( using npm install )
  - ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

  - Device: Macbook Pro 16" (2020)
  - OS version: 10.15.7

    - ( ) Windows : version : 
    - ( ) Linux : version : 
    - (x ) MacOs : version : 10.15.7
    - ( ) Raspbian: version :
    - ( ) Other : specify :

  - Description of the other OPCUA system I am trying to connect to:

node-opcua Client connecting to node-opcua Server
version 2.29.0 (pre-release)

- **node-opcua version:** :
   v2.29.0 (pre-release) although its latest release through `npm update`

  
  

- **Node:**
  `node --version` = v15.5.0

erossignon, 2021-01-14:
Try to remove  the  "/Users/srogers/Library/Preferences/NodeOPCUA-Default-nodejs" so that certificate get recreated ..

srogers2112, 2021-01-18:
Removing the the folder fixed the issue.

closed 2021-01-18 by srogers2112

jumo-sballing, 2021-01-25:
Same problem here on an embedded ARM device with linux 5.4.44, node v12.14.1, node-opcua 2.30.0. Deleting the folder NodeOPCUA-Default-nodejs results in automaticly regeneration of this folder. But the error still remains.

Should I reopen this issue or generate an new one?

jumo-sballing, 2021-01-25:
I'm sorry. My fault. With version 2.31.0 the error is gone. Sorry for interruption. Case closed.

paulsalibe, 2021-03-17:
I am using 2.113 and this is still happening. Delete the folder and it comes right back immediately.