node-opcua/node-opcua #931: How remove event "backoff" when disconnect the client?

mariodiciccio, 2021-01-14:
Hi,
my question is this: the client is not connect to LAN, but i try to connect through the method client.connect(). the event "backoff" trigger continually until the client is connected.

if i use client.disconnect() the event is not removed.

How disconnect it completely?

Thanks

erossignon, 2021-01-14:
Can you provide a gist that isolate and highlight the issue ?

erossignon, 2021-01-17:
see  #531

mariodiciccio, 2021-01-18:
Thanks for reply, but i do not want to remove this event, but when disconnect the client the "backoff" event trigged again.

this.client = opcua.OPCUAClient.create();
  this.client.on("backoff", function() {
            console.log("still trying to connect to ", self.endPoint ,": retry =", retry, "next attempt in ", delay/1000, "seconds" );
         });

this.client.disconnect(function(){
              console.log('disconnected ');
});

erossignon, 2021-01-19:
OK, seems to be a edge case, when/how are you calling connect ?

mikakaraila, 2021-01-19:
Just a comment, if you want to change connection to another endpoint then code must remove first old listeners before connecting to another server. I just added this functionality to node-red-contrib-opcua. Take a look into 102-opcuaclient.js.

There is reconnect(msg) function that contains removeListener() use.

mariodiciccio, 2021-01-24:
I will follow your advice.

thank you very much

erossignon, 2021-01-25:
node-opcua@2.31.0 fixes the event "backoff" that may still be raised after client#disconnect has been called.
@mariodiciccio   could you confirmed that the issue is now solved on your end with node-opcua@2.31.0  ?

mariodiciccio, 2021-01-30:
its seems well

erossignon, 2021-01-30:
Grwat. Thanks

closed 2021-01-30 by erossignon