node-opcua/node-opcua #936: Unhandled Rejection while server certificate checks

bruegth, 2021-01-19:
**Current behavior**

```
08:27:58.808Z :verify                        :46   [NODE-OPCUA-W04] Warning: the certificate status is =  BadSecurityChecksFailed  file =  /certs/server_certificate.pem
Unhandled Rejection at Promise: Message: Cannot read property '0' of undefined, Stack: TypeError: Cannot read property '0' of undefined
    at OPCUAServer.<anonymous> (\node_modules\node-opcua-client\dist\verify.js:77:303)
    at Generator.next (<anonymous>)
    at fulfilled (\node_modules\node-opcua-client\dist\verify.js:5:58)
```

**Describe the bug**
If the Server certificate is not valid, a unhandled Rejection is thrown.

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua from source ( using git clone)
  - (X ) I have installed node-opcua as a package ( using npm install )
  - ( ) I am using an application that uses node-opcua

- **node-opcua version:** : 2.30.0

- **Node:**
  `node --version` =14.8.0

erossignon, 2021-01-23:
Thank Thomas, it should not raised this unhandled exception. 
The certificate verification code that we recently introduced to check that your server' certificate is correct has highlight a conformance issue (in the certificate) that we didn't accounted for properly.

erossignon, 2021-01-24:
It should be fixed in 2.31 ,please confirm

bruegth, 2021-01-25:
I can confirm that this is fixed with 2.31.0

closed 2021-01-25 by bruegth