node-opcua/node-opcua #937: Loading enums from nodeset.xml files does not work for enums with negative values

Magnulas, 2021-01-19:
**Current behavior**

When loading a nodeset from a xml file, and the xml file contains a enum with negative values a error will be thrown, saying "Error: Invalid Extension Object". See attached project/zip for a example of the error.

**Describe the bug**

In part 6 of the OPC UA specification, section 5.2.4, enumrations are defined as Int32 values:

_Enumerations are encoded as Int32 values._

As such it should be able to handle negative values up to -2147483648.

**To Reproduce**

Load a nodeset xml with a enum which has a value in the range [-2147483648,-1]. See attached project/zip for a running example.

**Expected behavior**

The nodeset should load successfully, as negative values in the range [-2147483648,-1] should be allowed according to the OPC UA specification.

**Gist**
node-opcua does not handle enumerations as Int32 values, but instead as UInt32, when the enum definition is loaded from a nodeset xml file

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua from source ( using git clone)
  - (X ) I have installed node-opcua as a package ( using npm install )
  - ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

  - Device: PC
  - OS version: Windows 10

    - (X ) Windows : version : 10.0.18363 Build18363
    - ( ) Linux : version : \***\*\_\*\***
    - ( ) MacOs : version : \***\*\_\*\***
    - ( ) Raspbian: version : \***\*\_\*\***
    - ( ) Other : specify :

- **node-opcua version:** :
2.30.0

- **Node:**
  `node --version` = v14.15.1
[NodeOpcuaError.zip](https://github.com/node-opcua/node-opcua/files/5836186/NodeOpcuaError.zip)

erossignon, 2023-06-03:
was fixed in node-opcua@2.66.0

closed 2023-06-03 by erossignon

erossignon, 2023-06-08:
definitely fixed and verified in node-opcua@2.104.0