node-opcua/node-opcua #938: Is there any support to browse the AddressSpace of server and generate nodeset file ?

Graos, 2021-01-21:
**Current behavior**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1. ....
2. ......
   3.......

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua from source ( using git clone)
  - ( ) I have installed node-opcua as a package ( using npm install )
  - ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

  - Device: **\_**
  - OS version: **\_**

    - ( ) Windows : version : \***\*\_\*\***
    - ( ) Linux : version : \***\*\_\*\***
    - ( ) MacOs : version : \***\*\_\*\***
    - ( ) Raspbian: version : \***\*\_\*\***
    - ( ) Other : specify :

  - Description of the other OPCUA system I am trying to connect to:

    - Name:\***\*\_\*\***
    - Version:\***\*\_\*\***
    - Manufacturer/Software vendor:\***\*\_\*\***
    - link : https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

MarcoSeb, 2021-01-21:
generate xml nodeset file:: http://node-opcua.github.io/api_doc/2.20.0/classes/uanamespace.html#tonodeset2xml

Graos, 2021-01-21:
I tried this in server it works fine. How do i do from client perspective ?

closed 2021-01-22 by Graos

simonbowen, 2021-03-17:
@Graos I am also looking for this answer, did you manage to generate a nodeset using client instead of a server?

Graos, 2021-03-17:
No I did not manage to generate a nodeset file from Node OPC UA client. In fact I doubt if there is any direct interface to do that. To my understanding client needs to browse the server and generate a nodeset file from those responses based on the schema defined in part 6.

simonbowen, 2021-03-17:
Ah OK. That's a real shame, I need to simulate a server and was hoping to export it to XML.

I am having to manually build it which is cumbersome, thank you for your response.

mikakaraila, 2021-03-17:
Server has dumpXML() but not really tested / used. If you write your server with node-opcua then you could add method to call it.
And do some testing.
BTW UaExpert has Document for nodeset that can export XML, but I have not tested it neither.

simonbowen, 2021-03-17:
@mikakaraila Thanks for that information. Unfortunately that isn't what I am trying to achieve. I want to connect to a 3rd Party server and dump the nodes to XML. This means I need to use the client, not define my own server.

mikakaraila, 2021-03-17:
Have you tried with UaExpert:
[image]

simonbowen, 2021-03-18:
@mikakaraila I have tried with UAExpert, with the free license it only supports exporting upto 20 nodes.

mikakaraila, 2021-03-18:
Sorry forgot that limitation