node-opcua/node-opcua #940: Method with user permission set is not working in 2.30.0 but it works fine in 2.28.3

Graos, 2021-01-21:
**Current behavior**

Methods with permission set like below
                    node.setPermissions({ CurrentRead: ["!", "AuthenticatedUser", "SecurityAdmin", "Developer"], CurrentWrite: ["!", "SecurityAdmin", "Developer"] });

throws user access denied for the user with role USER1_ROLE": "Developer;SecurityAdmin"

**Describe the bug**
Methods with permission set like below
                    node.setPermissions({ CurrentRead: ["!", "AuthenticatedUser", "SecurityAdmin", "Developer"], CurrentWrite: ["!", "SecurityAdmin", "Developer"] });

throws user access denied for the user with role USER1_ROLE": "Developer;SecurityAdmin" This was working in 2.28.3 after updating to 2.30.0 imethod execution results in User access denied.

**Gist**
gist:

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua as a package ( using npm install )
  
    - ( ) node-red
    - ( ) other : please specify

  - Device: **\_**
  - OS version: **\_**

    - ( ) Windows : version : \***\*\_\*\***
    - ( ) Linux : version : \***\*\_\*\***
    - ( ) MacOs : version : \***\*\_\*\***
    - ( ) Raspbian: version : \***\*\_\*\***
    - ( ) Other : specify :

  - Description of the other OPCUA system I am trying to connect to:

    - Name:\***\*\_\*\***
    - Version:\***\*\_\*\***
    - Manufacturer/Software vendor:\***\*\_\*\***
    - link : https://

- **node-opcua version:** : 2.30.0
  
  

- **Node:**
  `node --version` =

MarcoSeb, 2021-01-21:
change ! => !* `node.setPermissions({ CurrentRead: ["!*", ...`

Graos, 2021-01-21:
But i want to provide method access only to the users having user role either SecurityAdmin or Developer.

Graos, 2021-01-21:
Sorry its my mistake. I should have included the roles for set permissions in the Execute property for methods.

closed 2021-01-21 by Graos