node-opcua/node-opcua #941: UaExpert throws BadCertificateHostNameInvalid when server used with default certificate

Graos, 2021-01-22:
**Current behavior**

UaExpert throws BadCertificateHostNameInvalid when server used with default certificate

**Describe the bug**
NodeOPCUA server uses the default certificate created in %APPDATA%. when the client(UaExpert') connects to server with Basic256Sha256 (sign and Encrypt) UaExpert throws BadCertificateHostNameInvalid(Hostname or IP address in certificate does not match with the hostname or IP address the client connected to)

**To Reproduce**
Steps to reproduce the behavior:

1. .Run the NodeOPCUA server
2. .connect the client (UaExpert) to server with Basic256Sha256 security policy with sign and encrypt)
   3.......

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Gist**
gist:

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

  - ( ) I have installed node-opcua as a package ( using npm install )

    - ( ) node-red
    - ( ) other : please specify

  - Device: **\_**
  - OS version: **\_**

    - ( ) Windows : version : \***\*\_\*\***
    - ( ) Linux : version : \***\*\_\*\***
    - ( ) MacOs : version : \***\*\_\*\***
    - ( ) Raspbian: version : \***\*\_\*\***
    - ( ) Other : specify :

  - Description of the other OPCUA system I am trying to connect to:

    - Name:\***\*\_\*\***
    - Version:\***\*\_\*\***
    - Manufacturer/Software vendor:\***\*\_\*\***
    - link : https://

- **node-opcua version:** : 2.30.0
  
  

- **Node:**
  `node --version` =

AndreasHeine, 2021-01-22:
have you renamed your computer after installing UaExpert (the certs are generated while installation) you can look inside the cert:
UaExpert -> Settings -> Manage Certificates -> <Trusted Own Certificate> -> open it -> lookup DNS-Name compare it against your computername!

Graos, 2021-01-22:
Domain name from server certificate in UaExpert shows  DE-L-7224240 and the endpoint to connect to is DE-L-7224240.EUROPE.ABB.COM. The difference is what the UaExpert complains about.

AndreasHeine, 2021-01-22:
add the FHQN to the UaExperts own cert under domainnames like i mentioned in "FYI"!

[image]

[image]

AndreasHeine, 2021-01-22:
if you create your certs with open ssl you could use the ip instead of the domain name

closed 2021-01-22 by Graos