node-opcua/node-opcua #942: d.ts files in 2.30 in uppercase, but imports are lowercase - tsc does not compile

Schloemicher, 2021-01-22:
**Current behavior**

Error TS2307: Cannot find module './i_register_server_manager' or its corresponding type declarations.

**Describe the bug**
node_modules/node-opcua-server/dist/register_server_manager.d.ts:6:40
> import { IRegisterServerManager } from "./i_register_server_manager";

File in dist:
...
I_register_server_manager.d.ts
...

in the package on NPM, the d.ts file is in uppercase since 2.30, but because imports are case sensitive it doesnt work anymore

**Reproduce**
1. Install node-opcua v2.29 -> works
2. Upgrade to node-opcua v2.30 -> does not compile

**Expected behavior**

change import to uppercase or release .d.ts files in lowercase

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
  - (x) I have installed node-opcua as a package ( using npm install )

  - OS version: 
    - (x) Linux : version : Ubuntu 20.4.1

- **node-opcua version:** : 2.30.0
- **typescript version:** : 4.1.3

- **Node:**
  `node --version` = 15.6.0

erossignon, 2021-01-23:
Good catch.. should be fixed in next patch..

erossignon, 2021-01-24:
Fixed in 2.31

closed 2021-01-24 by erossignon