node-opcua/node-opcua #944: monitoredItem.on is not function

hmthang396, 2021-01-30:
Hi,
My error is this: monitoredItem.on("changed",(dataValue) => {
                      ^
TypeError: monitoredItem.on is not a function
I'm try to read web value changed by it can't Although I follow like "sample_client.js" example

const opcua = require("node-opcua");
const async = require("async");
const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: 1
};
const options = {
    applicationName: "MyClient",
    connectionStrategy: connectionStrategy,
    securityMode: opcua.MessageSecurityMode.None,
    securityPolicy: opcua.SecurityPolicy.None,
    endpoint_must_exist: false,
};
const client = opcua.OPCUAClient.create(options);
const endpointUrl = "opc.tcp://xxx.xxx.xxx.xxx:4840";
client.on("backoff", (retry, delay) => 
    console.log("still trying to connect to ", endpointUrl ,": retry =", retry, "next attempt in ", delay/1000, "seconds" )
);
let the_session, the_subscription;
async.series([
    // step 1 : connect to
    function(callback)  {
        client.connect(endpointUrl,function(err){
            if(err)
            {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            }else{
                console.log("connected !");
            }
            callback(err);
        });
    },
    // step 2 : createSession
    function(callback) {
        client.createSession(function(err,session){
            if(err){
                return callback(err);
            }
            the_session = session;
            callback();
        });
    },
    // step 3 : browse
    function(callback) {
        the_session.browse("RootFolder", function(err, browseResult) {
            if(!err) {
                console.log("Browsing rootfolder: ");
                for(let reference of browseResult.references) {
                    console.log( reference.browseName.toString(), reference.nodeId.toString());
                }
            }
            callback(err);
        });
    },
    // step 4 : read a variable with readVariableValue
    function(callback) {
        const maxAge = 0;
        const nodeToRead = { 
            nodeId: "ns=4;s=GVL_Value.LEVEL_HIGH", 
            attributeId: opcua.AttributeIds.Value
        };
        the_session.read(nodeToRead, maxAge, function(err, dataValue) {
            if (!err) {
            console.log(" Value = " , dataValue.value.value);
            }
            callback(err);
        });
    },
    // step 4' : read a variable with read
    function(callback){
        the_session.readVariableValue("ns=4;s=GVL_Value.LEVEL_HIGH", function(err, dataValue) {
            if (!err) {
                console.log(" Value = " , dataValue.value.value);
            }
            callback(err);
        });
    },
    // step 5: install a subscription and install a monitored item for 10 seconds
    function(callback) {
        //"install a subscription"
        const subscriptionOptions = {
            maxNotificationsPerPublish: 1000,
            publishingEnabled: true,
            requestedLifetimeCount: 100,
            requestedMaxKeepAliveCount: 10,
            requestedPublishingInterval: 1000
        };
        
        the_session.createSubscription2(subscriptionOptions, (err, subscription) => {

            if(err) { return callback(err); }
        
            the_subscription = subscription;
        
            the_subscription.on("started", () => {
                console.log("subscription started for 2 seconds - subscriptionId=", the_subscription.subscriptionId);
            }).on("keepalive", function() {
                console.log("subscription keepalive");
            }).on("terminated", function() {
               console.log("terminated");
            });
            callback();
        });
        
    },
    
    function(callback) {
        
        //"add some monitored items"
        const itemToMonitor = {
            nodeId: opcua.resolveNodeId("ns=4;s=GVL_Value.VC_1_NOW"),
            attributeId: opcua.AttributeIds.Value
        };
        const monitoringParamaters = {
            samplingInterval: 100,
            discardOldest: true,
            queueSize: 100
        };
        const monitoredItem =the_subscription.monitor(itemToMonitor,monitoringParamaters,opcua.TimestampsToReturn.Both);
        console.log("-------------------------------------");
        monitoredItem.on("changed",(dataValue) => {
            console.log(dataValue.value.toString());
        });

    },
    function(callback) {
        // wait a little bit : 10 seconds
        setTimeout(()=>callback(), 10*1000);
    },
    // terminate session
    function(callback) {
        the_subscription.terminate(callback);;
    },
    function(callback) {
        //"closing session"
        the_session.close( function(err) {
            if(err) {
                console.log("closing session failed ?");
            }
            callback();
        });
    }
],
    function(err){
    if(err)
    {
        console.log(" failure ",err);
    }else
    {
        console.log("done!");
    }
    client.disconnect(function(){});
});

[image]

erossignon, 2021-01-30:
You code differs from the https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_client.js 

````javascript
 the_subscription.monitor(
             itemToMonitor,
             monitoringParamaters,
             TimestampsToReturn.Both,
             (err, monitoredItem) => {
               monitoredItem.on("changed", function(dataValue) {
                 console.log(
                   "monitored item changed:  % free mem = ",
                   dataValue.value.value
                 );
               });
               callback();
             }
           );
```` 

Also, we recommand the async/await technic rather than the callback API to get clearer code; check https://leanpub.com/node-opcuabyexample

closed 2022-02-27 by erossignon

pontarolli, 2024-02-02:
I had the same problem when trying to run the example sample_client.js [v2.1.3](https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/sample_client_ts.ts) 

Part of the original sample_client.js v2.1.3 that was causing the error  

```js
function(callback) {
    // install monitored item
    const monitoredItem  = the_subscription.monitor({
            nodeId: opcua.resolveNodeId("ns=1;s=free_memory"),
            attributeId: opcua.AttributeIds.Value
        },
        {
            samplingInterval: 100,
            discardOldest: true,
            queueSize: 10
        },
        opcua.TimestampsToReturn.Both
    );
    console.log("-------------------------------------");

    monitoredItem.on("changed", function(dataValue) {
        console.log("monitored item changed:  % free mem = ", dataValue.value.value);
    });
},
```
    

Modified function of the previously posted function that corrected the error
```js
function(callback) {
    const itemToMonitor = {
        nodeId: opcua.resolveNodeId("ns=1;s=free_memory"),
        attributeId: opcua.AttributeIds.Value
    };

    const monitoringParamaters = {
        samplingInterval: 100,
        discardOldest: true,
        queueSize: 10
    };

    const monitoredItem = the_subscription.monitor(
        itemToMonitor,
        monitoringParamaters,
        opcua.TimestampsToReturn.Both,
        function (err, monitoredItem) {
            if (err) {
                console.error("Error creating monitored item:", err.message);
                return callback(err);
            }

            console.log("-------------------------------------");

            monitoredItem.on("changed", function (dataValue) {
                console.log("monitored item changed: % free mem =", dataValue.value.value);
            });

            callback();
        }
    );
},
```

Hope this helps!