node-opcua/node-opcua #947: Unable to Browse the Functional Blocks created in TIA Projects.

shribadiger, 2021-02-10:
**Current behavior**

We have a Functional Blocks in the TIA Projects running in Siemens 1500 PLCs. Node OPCUA Client is able to browse the tags which are present inside the data block but its not able to browse the tags inside the functional block.

**Describe the bug**
We are using the node-opcua-client to browse the tags from the Siemens PLCs 1500. We are getting the configured tags in the plain data block. But if we create the functional blocks, not able to browse the tags and giving wrong address format. Screen shot of address format is attached. As of now, We are using the docker to run node-opcua clinet hence not able to collect the debug logs. Need support to collect logs in docker services(debug logs).
[image]

Browse Tags: V_"FB_1_Sequence"."ExternalSignal"  -- Wrong address to read tag from the PLCs via OPCUA Channel.

**To Reproduce**
1. Configure the functional blocks in TIA ( Siemens Total Integrated Automation Tool)
2. Start OPCUA Server in PLC 1500 Family
3. Start to browse the tags by node-opcua-client by passing the IP of 

**Expected behavior**
It should browse the Functional Blocks and display proper Tag Address. 
**Screenshots**
If applicable, add screenshots to help explain your problem.

  - Device: Siemens S7 1500 PLCs
  - OS version: Debian 10

- **Node:**
  `node --version` =Node 12 version.

AndreasHeine, 2021-02-10:
which node-opcua version? 
working alot with s7-1500 opcua servers i'll make a quick test! just one more question: can you browse the tags with UaExpert?

AndreasHeine, 2021-02-10:
no issues...
CPU 11505SP F FW2.6
node-opcua 2.33.0
[image]

but check if the tags are avalible with UaExpert! if not it might be a wrong OPC UA Configuration in the s7 or sometimes you need to reload the Hardware-Configuratiration if you changed OPC UA Server relevant stuff in the plc program.

closed 2022-04-16 by stale[bot]